PX4/PX4-Autopilot #9452: Cannot get a green light with LPE with poses provided by a SLAM

FaboNo, 2018-05-12:
Dear All,

I am trying to control a quadrotor with poses provided by a SLAM in a GPS denied environment (indoor). But before going OFFBOARD, I want to be sure that the PX4 is taking into account the poses sent by the SLAM , and thus normally after arming the quadrotor I should have a Solid green light and if I switch to POSCTL, it should remains Solid green.

**Intel onboard NUC**

- Telem2 with UART 921600
-  Ubuntu 16.04
-  ROS Kinetic
- Current MAVROS Branch

**QgroundControl V3.3.1**

**Pixracer** 

I am using a pixracer and  the latest stable version v1.7.3 downloable from QGC) – it includes LPE and the Attitude_Q_estimator.

Regarding Attitude Q estimator : here is the setup :

ATT_EXT_HDG_M set to « vision »

[image]

As we are in a denied GPS environment I disabled all GPS compensations.

Regarding LPE here is the setup :

SYS_MC_EST_GROUP  set to LPE & attitude estimator
LPE_fusion – fuse vision position and baro
[image]

To enable LPE to initialize itself : LPE_FAKE_ORIGIN = 1 

For my tests, I do not have RC, so :

NAV_RCL_ACT=0 (disabled in order to avoid failsafe mode enabled)  

COM_RC_IN_MODE = 1 (Joystick/no RC checks)

Regarding GPS :

CBRK_GPSFAIL = 240024 in order for the Pixhawk to ignore checking whether I have GPS receiver or not.

The SLAM is on and sends positions via /mavros/vision_pose/pose

If I take a look at M1:LOCAL_POSITION_NED.x/y/z, it seems that LPE is working fine and takes into account the pose sent by the SLAM. When I am moving by hands the quadrotors, I can see that the same moves in the NED frame.

Via mavlink the « commander status » command shows me that the correct local pose is taken into account :
[image]

I noticed that FCU thru MAVROS is still keep saying :

GP : NO GPS FIX
TM : Wrong FCU time

But I do not have « HP : Requesting home position » anymore (I guess because LPE_FAKE_ORIGIN = 1)

I know that whose are warnings but I am wondering how can I fix the « wrong FCU time » issue ? From my understanding, it is related to the fact that I do not have GPS position – can somebody is able to confirm ?

However after arming, I still a have a Solid blue light on the pixracer LED (ARM, NO GPS according to the px4 documentation), it does not switch to solid GREEN, even though I switch to POSCTL, meaning that there is no position lock, according to px4 documentation (https://dev.px4.io/en/ros/external_position_estimation.html).

Can somebody is able to tell me why and what is wrong and/or missing in my setup ?

Thank you very much

TSC21, 2018-05-12:
@FaboNo even if the LED does not turn green, are you able to get `LOCAL_POSITION_NED` msgs on QGC?

FaboNo, 2018-05-12:
@TSC21 yes when I move the drone, I can observe the same change on LOCAL_POSITION_NED (but not on the same axis because the SLAM is working in ENU frame)

TSC21, 2018-05-12:
That means that you have position lock but somehow there's a bug on the system reporting that the position lock is working. That needs to be debugged on that sense, but you probably still are able to change to POSCTL. Have you tried it?

FaboNo, 2018-05-12:
Yes, I can switch on POSCTL (rosrun mavros mavsys mode -c POSCTL) and the light remains solid BLUE.

TSC21, 2018-05-12:
Yes, so that's a bug. Have to check the problem. Thanks for reporting.

FaboNo, 2018-05-14:
I am wondering whether I do not get the green light because I do not have RC  (although iI set NAV_RCL_ACT=0). I will test with RC as soon as possible and let you know what happen.