PX4/PX4-Autopilot #9453: Can I turn on two UAVCAN processes at the same time?

BreederBai, 2018-05-13:
I now want to make some interesting changes to the PX4 so that he can convert the received UART, SPI, or I2C signals to standard UAVCACN messages. However, I encountered some difficulties in realizing this idea. If I want to forward two UART GPS messages at the same time, then the UAVCAN GNSS message I forwarded should have two NODE ID, but at present there is only one NODE ID for a UAVCAN process. I would like to solve this problem by enabling two UAVCAN processes with different NODE ID. Is this method theoretically feasible?

dagar, 2018-05-13:
I'm sure you could hack this if truly necessary, but I'd question the need. Could you do something more interesting with the data first?

For example what if you blended the multiple GPS readings and only published the result over CAN?

If you also have an IMU you could run the estimator and publish the entire position estimate. https://github.com/UAVCAN/dsdl/blob/master/uavcan/navigation/2000.GlobalNavigationSolution.uavcan

Personally I don't see the point of using a CAN bus like a dumb pipe for data. You already have a fast processor directly connected to the sensor. What can you do with it?

BreederBai, 2018-05-14:
@dagar Thank you very much for your answer! The reason why I want to do this is because I want to use the UAVCAN protocol, but currently the devices that support the UAVCAN protocol are very rare. So I want to build a signal conversion device based on the PX4 architecture.
If two UAVCAN processes are enabled at the same time. My biggest doubt is that both processes use CAN hardware to send and receive information. Will these two processes affect each other?

LorenzMeier, 2018-05-14:
Given your interest in writing a translation layer I believe that you need to dig into the code anyway and on the lower levels you can certainly mix CAN with UAVCAN.

closed 2018-05-14 by LorenzMeier

BreederBai, 2018-05-14:
@LorenzMeier Thank you very much for your answer! I have been watching UAVCAN part code recently. However, there are differences in the implementation of UAVCAN between ardupilot and PX4, which makes me fall into a dilemma. When ardupilot creates a node, the incoming parameter is a specific CAN hardware pointer, and when PX4 creates the node, the incoming parameter is a collection of CAN hardware pointers. I think as you said, I still need to dig into this part of the code.