PX4/PX4-Autopilot #9455: How can I build 1.7.4beta?

jstrebel, 2018-05-14:
I tried to build a 1.7.4beta, because I get in QGC 3.2.2 error messages when I upload the latest 1.8 test version into the Pixhawk.
So I downloaded 1.7.4beta and tried to build it in the Directory /Firmware-1.7.4beta. 
I get the error message "You have to use GIT to download this repository".
Stupid question. How I can I do this?
Thanks for advises. Jakob

mcsauder, 2018-05-15:
Hi @jstrebel , You should start here: https://dev.px4.io/en/setup/getting_started.html

jstrebel, 2018-05-17:
This way worked for me.
## Cloning the Repository:
git clone https://github.com/PX4/Firmware.git
cd Firmware
## List the available tags:
git tag –l
## checkout the desired tag (i.e 1.7.4beta)
git checkout -b tags/v1.7.4beta
## Build the desired version:
make px4fmu-v2_default

closed 2019-02-06 by stale[bot]