PX4/PX4-Autopilot #9461: GPS not found after boot, need to Stop/Start

tops4u, 2018-05-14:
**Bug Report**

My ublox 8M GPS is most of the time not found after boot. Sometimes Powercycling helps or issuing a GPS Stop / GPS Start command. Before the GPS is cycling various GPS Products but is not able to find the ublox. Is this HW related - I use a mRo X2.1. 

[image]

Bootlog below:
`[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1

[boot] Fault Log is Armed

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: AUAV_X21
FW git-hash: 6b94ef1a03796cd8378a4948bde2f36db7527ec6
FW version: 1.8.0 80 (17301632)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ac630e6d0e90480121c74d59a92676f0b951dce
Build datetime: May  4 2018 21:56:47
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 363633313236510d004b002b
MCU: STM32F42x, rev. 3
UID: 4B002B:3236510D:36363331 
[hardfault_log] Fault Log is Armed

INFO  [tune_control] Publishing standard tune 1
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
nsh: rgbled_pwm: command not found
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
WARN  [lis3mdl] no device on bus 2
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 1 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
MPU9250 on SPI bus 1 at 4 (1000 KHz)
INFO  [mpu9250] accel cutoff set to 30.00 Hz
INFO  [mpu9250] gyro cutoff set to 80.00 Hz
ERROR [sf1xx] driver start failed
nsh: send_event: command not found
INFO  [load_mon] stack check enabled
ERROR [param] Parameter CAM_FBACK_MODE not found.
nsh: camera_feedback: command not found
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: Onboard, data rate: 5000 B/s on /dev/ttyS2 @ 57600B
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Mixer: /etc/mixers/hexa_x.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 
Addons script: /fs/microsd/etc/extras.txt 
INFO  [mavlink] mode: OSD, data rate: 10000 B/s on /dev/ttyS6 @ 57600B
WARN  [mavlink] hardware flow control not supported
INFO  [logger] logger started (mode=all)`

tops4u, 2018-05-14:
This Problem arose between 1.7.3 and 1.7.4 as far as I can tell. 1.7.3 behaves ok, whereas 1.7.4 and newer exposes the above behavior.

The only change during this time I found on GPS was https://github.com/PX4/Firmware/commit/03b8cd78b3e32be581f5e9ecf22358da03b3fe91#diff-99edb50b3a54c5fd16d92e7fb53eecd5 but CTS/RTS is not used and UART4 (the one in use) does not have CTS/RTS anyway. Strange though, this may be a mRo X2.1 Issue as the Port in use is UART4/ttyS3 which is per default not mapped (as per init.d/rcS):

```
# UART mapping on FMUv2/3/4:
#
# UART1			/dev/ttyS0		IO debug (except v4, there ttyS0 is the wifi)
# USART2		/dev/ttyS1		TELEM1 (flow control)
# USART3		/dev/ttyS2		TELEM2 (flow control)
# UART4
# UART7							CONSOLE
# UART8							SERIAL4
```

bkueng, 2018-05-15:
My first guess would have been https://github.com/PX4/Firmware/commit/03b8cd78b3e32be581f5e9ecf22358da03b3fe91#diff-99edb50b3a54c5fd16d92e7fb53eecd5 as well.
Can you do a `git bisect`?

> as the Port in use is UART4/ttyS3 which is per default not mapped (as per init.d/rcS):

It's (incorrectly) just not mentioned there, but ttyS3 is the default GPS port.

tops4u, 2018-05-15:
@bkueng I'll do that.

bkueng, 2018-05-16:
hmm that is strange. The diff between 873e036 and 8a114c2f050ccf5d5 is small and it does not look as if any of the changes could be related.
But just to be sure, did you try to build 8a114c2f050ccf5d5 yourself?
Did you repeat the test to exclude some random behavior?
One more thing is that the binary from QGC uses a much newer GCC version (7.2.1). This could very well be the cause.

tops4u, 2018-05-16:
@bkueng 
I would need a hint on how to do the GIT Stunt (I'm no git Guru). Tried git reset --hard <id> but this did not work. 
Besides I don't think it is related to this code change, as it also fails with the 1.7.4 Version exactly the same way as Master. 
Yes I have repeated it several times on different FMU HW (using same GPS though, but I don't think it is GPS related). It is consistently failing on 1.7.4 and newer when installed with QGC and it is always working immediately if locally built or prior to 1.7.4.
I would be very much surprised if this is caused by the Buildchain, and why it would then only happen to some People and not all HW based on the same FMU Revision.

bkueng, 2018-05-17:
> I would need a hint on how to do the GIT Stunt (I'm no git Guru). Tried git reset --hard but this did not work.

`git checkout 8a114c2; git submodule update --recursive`

I just checked:
v1.7.3 was using GCC 5.4 (https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2)
v1.7.4beta is using GCC 7.2.1 (https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/7-2017q4/gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2)

tops4u, 2018-05-18:
I did some testing on this and it left me with more confusion...

Took another UBX (M8Q) with the mRo Unit and it behaves identically as with the M8N. Tried then on a Dropix where it worked in all conditions, but then this may have been because of the difference of the Build Target. I took my older AUAV-X2 Unit and tried the same (this one is using the Default FMUv2 Target), and this behaved the same as the mRo Units. 

So I tried to find differences on the HW Schematics. Between the original Pixhawk (FMUv2) and the mRo Unit there is a difference on the UART4: The Original is a 3V3 <-> 5V Levelshifter, that was left out on the mRo, AUAV but Dropix as well... As the Dropix seems similar from the HW Perspective as the mRo/AUAV I really don't know why they behave so different with identical Firmware.

bkueng, 2018-05-24:
@tops4u thanks for providing me with the HW. I was able to reproduce it and this is what I found so far:
- the toolchain seems to affect the problem, but it seems not to be the cause
- moving the `gps start` to the end of the startup script resolves/works around the issue
- reopening the uart port helps too. Using this diff:
```diff
diff --git a/src/drivers/gps/gps.cpp b/src/drivers/gps/gps.cpp
index f8755d8bc7..78a9328f55 100644
--- a/src/drivers/gps/gps.cpp
+++ b/src/drivers/gps/gps.cpp
@@ -788,6 +795,11 @@ GPS::run()
                 case GPS_DRIVER_MODE_ASHTECH:
                     _mode = GPS_DRIVER_MODE_UBX;
                     usleep(500000); // tried all possible drivers. Wait a bit before next round
+                    close(_serial_fd);
+                    _serial_fd = ::open(_port, O_RDWR | O_NOCTTY);
+                    if (_serial_fd < 0) {
+                        //...
+                    }
                     break;

                 default:
```
- When it fails to detect the GPS, this is what happens: the first few [reads](https://github.com/PX4/Firmware/blob/master/src/drivers/gps/gps.cpp#L419) return data, but then the poll() always times out, and `ioctl(_serial_fd, FIONREAD, (unsigned long)&bytesAvailable);` indicates there is no data available. Looks like a lower-level issue.

@davids5 any ideas?

davids5, 2018-05-24:
@bkueng - I am trying to relate the late start of the gps driver fixing the problem.....

I would scope the interface and ensure the power and signals are clean. Ensure the power is not having a slow start or being toggled at boot. 

>the toolchain seems to affect the problem, but it seems not to be the cause
what do you mean by affect but does not cause?

There is a high possibility it is memory corruption startup order would affect the memory layout. 

Add a startup script to override the one in FLASH. In it just start the GPS. Does it work every reboot?
That will eliminate the OS/driver or another driver using the port. You could also check the open reference count (in the debugger) where you closed and reopened the port. Is it 1 before the close? If not something else is reading the port.

bkueng, 2018-05-29:
> what do you mean by affect but does not cause?

I don't have a clear picture on that yet. What I noticed is that with different toolchain versions, it's different how often the error occurs.

>  In it just start the GPS. Does it work every reboot?

Yes that works.

> You could also check the open reference count (in the debugger) where you closed and reopened the port

I checked this value: `tcb->group->tg_filelist.fl_files[_serial_fd].f_inode->i_crefs`. It is always 1, so no one else is using it. I also checked that `open` for `ttyS3` is only called from the `gps` driver.

I scoped the signal, this is the startup going into bad state:

[image]
The first 2 blocks are good transfers, but after that I never see any good signal anymore, until I restart the `gps` driver. Then the GPS gets detected and the signal looks good:
[image]

LorenzMeier, 2018-05-29:
Thanks for the investigation @bkueng

davids5, 2018-05-29:
@bkueng - Please add all the FMU serial interface signals to the capture. Please also confirm that the power was shut off at +3 Seconds on the first picture.

bkueng, 2018-05-29:
> Please add all the FMU serial interface signals to the capture

Do you mean TX?

> Please also confirm that the power was shut off at +3 Seconds on the first picture.

There was no power off at +3s. This was during normal power up. There was no reboot between picture 1 and 2, just `gps stop` and `gps start`.

philipoe, 2018-06-06:
Any updates on this? We can reproduce this on one of our platforms (PX4FMUv3, [this](https://store.mrobotics.io/mRo-GPS-u-Blox-Neo-M8N-HMC5983-Compass-p/mro-gps003-mr.htm) mRO GPS and PX4 Firmware at commit 25528a5ae0cc9cbada7c2daa266049f293fe05b0. I could retest any fixes/PRs that potentially solve this.

LorenzMeier, 2018-06-06:
@bkueng @davids5 Any insights?

bkueng, 2018-06-07:
@philipoe which board are you using exactly?

@LorenzMeier not yet. @davids5 wanted to reproduce it himself.
As I wrote above, we can work around the issue by reopening the UART on failed GPS scanning iterations. It's a simple way to unblock the release but it won't solve the root cause.

bkueng, 2018-06-07:
work-around is here: https://github.com/PX4/Firmware/pull/9614
@philipoe can you test it on your board?

tops4u, 2018-06-07:
Hi @bkueng thanks for the Progress in this case. I appreciate a Workaround but I think we should also figure out the root cause for this and why it is happening on some boards while it does not on others and why it has something to do with the Buildchain. 

It came to my mind, that the cause may be related to the Powersupply since AFAIK mRo and AUAV do have a switchable Supply (by Command from FMU) to the UART Port while Dropix for instance does not (always Powered with the FMU).

philipoe, 2018-06-07:
> @philipoe which board are you using exactly?
Pixhawk (1), but need to check whether from 3DR or mRo...
> @philipoe can you test it on your board?
Yes, but only on Monday...

davids5, 2018-06-07:
@LorenzMeier - I have tested this on a https://store.mrobotics.io/Genuine-PixHawk-1-Barebones-p/mro-pixhawk1-bb-mr.htm with a 3DR GPS. I am powered from a bench supply via the powermodule.  I am not able to reproduce. 

I will need to know the exact versions of hw that exhibits the failure.  Links to the FMU and the GPS would be super helpful.

pkocmoud, 2018-06-07:
@philipoe can you post your exact failing binary so that I can confirm on the same models of the hardware that I have here?

tops4u, 2018-06-07:
I did some investigation on this. 

@davids5 this is not a Pixhawk1 from mRo but the X2.1 Unit. I have sent my spare to Auterion, so my tests are run on a AUAV-X2 Unit that so far behaves identically as the mRo x2.1. 

I have attached a FTDI to the TX Line to check what is being sent to the GPS. (This is not readable, just to illustrate that there are obvious differences).

This is what happens on a "Homebrew" 1.7.3 during Boot GPS OK Afterwards:
```
   µb (    –   ñ  !         –   ñ  !      R,æ7    @‰ H` ‡ D ``  ∏ˇx<ˇ¯Ä x¿   x¿ x     ¯     x<Ä  x Ä x      ¯     ¯     x<Ä  x Ä x       x¿ ¯x<ˇ¯Ä x¿   x¿ x     ¯     Äx¿  x Ä x      ¯     ¯     Äx¿  x Ä x      ÄÄx< x¿µb »    ›hµb$$                                  \®µb Qµb Kµb 0 ;¢µb 
	pµb
  4

```

While this is what happens on Master from the Server Build GPS Unavailable afterwards:
`µb (    –   ñ  !         –   ñ  !      R,æ7     8(   4  Ñ Ñ`  „`

Reboot on 1.7.3 GPS Available afterwards:
```
µb (    –   ñ  !         –   ñ  !      R,æ7     8(   4  Ñ Ñ`  „x<ˇ¯Ä x¿   x¿ x     ¯     x<Ä  x Ä x      ¯     ¯     x<Ä  x Ä x       x¿ ¯fœ<x  ÄÜ      ˘ÄÄ   ûò              ˘ÄÄ   ûò         ‡ˇˇ*! "`"   †! 0 È$PMTK220,200*2C
$PGCMD,16,0,0,0,0,0*6A
$PMTK313,1*2E
$PMTK301,2*2E
$PMTK397,0*23
$PASHS,POP,20
$PASHS,NME,ZDA,3xòò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~0~00‡Ü0‡Ü˛òx0x0ÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~03`3‡Ü0‡Ü˛ò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü ô˛ò3‡Ü0‡Ü˛ò¯ò‡Ü á¯Ü áfÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~03x3‡Ü0‡Ü˛ò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü3 ô`0‡Ü0‡ÜÄò03`√ fÃf¿Ãf√f f0Ã‡Ü0‡ÜÜò03` ` ô03Äò3‡Ü ô˛ò ô‡Ü áÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Üòf0Ãò‚/qm *b≠\π"òc¶!∏£&!E‡%∞b¶„f™¶nqm *%â™π"Ä„¶!∏#@+Â¢PdV‡¶côÍV'ÄÍ±Í%!E‡%à$¶b`¢PdV‡¶%&™¶l  E‡%Ä#¶-%!E‡%∞b¶%èW—A=!äÙCq÷+Èzi1πŒ=!uÙ„\Cƒ#+Èzi1πŒ}—Bq÷–C∏Œ]%A=!"z˚q‘+iÇij qX§z˚qŒ+a=\q8πŒ=!™Ù#Cƒ#‘i qz1!‚#™±q‘+!Çi1πŒ=!äÙCqN+Èzi1πŒ=°¶!#\Cƒ#+Èzi1πŒ}—Bq÷–C¸µb (    –   ñ  !         –   ñ  !      R,æ7     r§@  ‡ D ``  ∏ˇx<ˇ¯Ä x¿   x¿ x     ¯     x<Ä  x Ä x      ¯     ¯     x<Ä  x Ä x       x¿ ¯fœ<x  ÄÜ      ˘ÄÄ   ûò              ˘ÄÄ   ûò         ‡ˇˇ*! "`"   †! 0 È$PMTK220,200*2C
$PGCMD,16,0,0,0,0,0*6A
$PMTK313,1*2E
$PMTK301,2*2E
$PMTK397,0*23
$PASHS,POP,20
$PASHS,NME,ZDA,ÇœÃò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~0~00‡Ü0‡Ü˛òx0x0ÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~03`3‡Ü0‡Ü˛ò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü ô˛ò3‡Ü0‡Ü˛ò¯ò‡Ü á¯Ü áfÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Ü~03x3‡Ü0‡Ü˛ò¯ò‡ÜÊÄòÄ` ô03Äò3‡Ü3 ô`0‡Ü0‡ÜÄò03`√ fÃf¿Ãf√f f0Ã‡Ü0‡ÜÜò03` ` ô03Äò3‡Ü ô˛ò ô‡Ü áÊÄòÄ` ô03Äò3‡Ü¯òÊòf0‡Üòf0Ãò‚/qm *%â™π"òc¶!∏£&!E‡%∞b¶„f™¶£qm *%â™π"Ä„¶!∏#@+Â¢PdV‡¶côÍV'ÄÍ±Í%!E‡%à$¶ `¢PdV‡¶%&™¶l  E‡%Ä#¶-%!E‡%∞b¶%ßW—Aä=!äÙCq‘+Èzi1πŒ=!uÙ#bCƒ#+Èzi1πŒ}—B£ –C∏Œ]%A=!"z˚qﬁ+iÇij qX§=q‘+a=\q8πŒ=!™Ù#Cƒ#‘i qz1!‚#™±qN+!Çi1πŒ=!äÙCqN+Èzi1πŒ=!uÙ#bCƒ#+Èzi1πŒ}—BqN–C¸
```

Reboot on 1.8 GPS unavailable afterwards:
`µb (    –   ñ  !         –   ñ  !      R,`

And this is what is happening when issuing a Stop - Start where the GPS starts working afterwards:
```
µb (    –   ñ  !         –   ñ  !      R,µb (    –   ñ  !         –   ñ  !      R,µb »    ›hµb$$                                  \®µb Qµb Kµb 0 ;¢µb 
	pµb
  4
```

tops4u, 2018-06-07:
@pkocmoud just get the latest Master via QGC and it should fail. For me 1.7.3 onwards fails.

LorenzMeier, 2018-06-07:
Latest master has a hotfix in.

tops4u, 2018-06-07:
@LorenzMeier thanks for pointing this out.
@pkocmoud so 1.7.3 from the Repo via QGC should also do the trick.

davids5, 2018-06-07:
>Any updates on this? We can reproduce this on one of our platforms (PX4FMUv3, this mRO GPS and PX4 Firmware at commit 25528a5. I tested with two GPS modules of the type mentioned above and had the same issue. I could retest any fixes/PRs that potentially solve this.

@tops4u:  @philipoe post seemed to indicate PX4FMUv3 exhibited the same problem, hence my test.  @pkocmoud is sending me an X2.1 and the mRo GPS mentioned. I will test this issue (without the work around) when I get that HW.

tops4u, 2018-06-08:
@davids5 my AUAV-X2 Unit is a FMUv2 and also exhibits this Problem. Intested with two different uBlox GPS from Drotek.

philipoe, 2018-06-11:
@bkueng Tested again, the Pixhawk used is the mRo version with "PX4FMU v2.4.6" printed on it. The fix seems to work, i.e. during 15 reboots it never completely failed to find the GPS. When issuing a "gps status" ca. 15 seconds after reboot, during 2 of the 15 reboots it first had not found the GPS, but then 2 seconds after the gps status had found it. See nsh output below (the test was done inside, which is why the lat/lon/alt values are not valid). But i assume this is normal and it just took it quite long to connect to the GPS.

> nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: OK
INFO  [gps] sat info: disabled
 vehicle_gps_position_s
        timestamp: 19783468 (0.130657 seconds ago)
        time_utc_usec: 0
        lat: 0
        lon: 0
        [...]

davids5, 2018-06-11:
@tops4u - I received AUAV X2.1 and the mRo GPS from mRo via @pkocmoud  (Thank you!).  I built and tested master @ 27e75920bd560b7d8de295db45aba8cf036cd98d with @bkueng 's reopen fix a62a71f48f3d40bba6e9af114a185f6bbad849b1 reverted. Building with gcc 5.4.1 or 7.2.1 gave me the same results. GPS status shows 

```
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: OK
INFO  [gps] sat info: disabled
 vehicle_gps_position_s
        timestamp: 11726380 (0.033118 seconds ago)
        time_utc_usec: 0
        lat: 0
        lon: 0
        alt: -17000
        alt_ellipsoid: 0
        s_variance_m_s: 57.210
        c_variance_rad: 3.142
        eph: 4294967.500
        epv: 3750001.500
        hdop: 99.990
        vdop: 99.990
        noise_per_ms: 98
        jamming_indicator: 38
        vel_m_s: 0.000
        vel_n_m_s: 0.000
        vel_e_m_s: 0.000
        vel_d_m_s: 0.000
        cog_rad: 0.000
        timestamp_time_relative: 0
        fix_type: 0
        vel_ned_valid: 0
        satellites_used: 0
INFO  [gps] rate position:                5.10 Hz
INFO  [gps] rate velocity:                5.10 Hz
INFO  [gps] rate publication:             4.74 Hz
INFO  [gps] rate RTCM injection:          0.00 Hz
```

So I am wondering what is different? 
I have a Power module connected.
Jtag and debug console.
USB cable (PC end disconnected_
mRo GPS only the 6 wire cable connected to X2.1 

[image]

What am I missing?

philipoe, 2018-06-11:
@davids5 How often did you test this (i.e. how many hard-reboots did you do)? For me, this did not happen after every reboot, but every 2nd or 3rd reboot.

davids5, 2018-06-11:
@philipoe At least 6 power ons and 8 reboot commands

tops4u, 2018-06-11:
@davids5 crosscheck with a build Binary from the Server. 1.7.3 is available for download it should be made with the new buildchain. It did not work when I tested it.

davids5, 2018-06-11:
@tops4u Please send me that .px4 file.

tops4u, 2018-06-11:
You can try this one. It's master from 16.5. (1.8. BETA).

[auav-x21_default.px4.zip](https://github.com/PX4/Firmware/files/2091106/auav-x21_default.px4.zip)

davids5, 2018-06-11:
@tops4u 
Interesting results. I had to load the binary with the uploader.  The bootloader on my AUAV X2.1 board reports the wrong PID and QGC (on windows) would not see the device in bootloader mode. 

```
Port2]  :  PX4 FMU
Device Power State:               PowerDeviceD0
       ---===>Device Information<===---
English product name: "PX4 BL FMU v2.x"
idVendor:                        0x26AC = 3D Robotics Inc.
idProduct:                       0x0011
bcdDevice:                       0x0101
iManufacturer:                     0x01
     English (United States)  "3D Robotics"
iProduct:                          0x02
     English (United States)  "PX4 BL FMU v2.x"
```

The start uptime seems to vary but it always started. I am repeatedly doing `gps status` every .25 Sec. on the dev console, I have seen it take 10 seconds.
```
NuttShell (NSH)
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 19200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> INFO  [Unknown] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
 gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: OK
INFO  [gps] sat info: disabled
 vehicle_gps_position_s
        timestamp: 10383451 (0.045011 seconds ago)
        time_utc_usec: 0
        lat: 0
        lon: 0
        alt: -17000
        alt_ellipsoid: 0
        s_variance_m_s: 20.928
        c_variance_rad: 3.142
        eph: 4294967.500
        epv: 3750000.250
        hdop: 99.990
        vdop: 99.990
        noise_per_ms: 104
        jamming_indicator: 14
        vel_m_s: 0.000
        vel_n_m_s: 0.000
        vel_e_m_s: 0.000
        vel_d_m_s: 0.000
        cog_rad: 0.000
        timestamp_time_relative: 0
        fix_type: 0
        vel_ned_valid: 0
        satellites_used: 0
INFO  [gps] rate position:                4.94 Hz
INFO  [gps] rate velocity:                4.94 Hz
INFO  [gps] rate publication:             0.12 Hz
INFO  [gps] rate RTCM injection:          0.00 Hz
nsh>  gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: OK
INFO  [gps] sat info: disabled
 vehicle_gps_position_s
        timestamp: 11183375 (0.104899 seconds ago)
        time_utc_usec: 0
        lat: 0
        lon: 0
        alt: -17000
        alt_ellipsoid: 0
        s_variance_m_s: 21.307
        c_variance_rad: 3.142
        eph: 4294967.500
        epv: 3750000.250
        hdop: 99.990
        vdop: 99.990
        noise_per_ms: 104
        jamming_indicator: 16
        vel_m_s: 0.000
        vel_n_m_s: 0.000
        vel_e_m_s: 0.000
        vel_d_m_s: 0.000
        cog_rad: 0.000
        timestamp_time_relative: 0
        fix_type: 0
        vel_ned_valid: 0
        satellites_used: 0
INFO  [gps] rate position:                4.94 Hz
INFO  [gps] rate velocity:                4.94 Hz
INFO  [gps] rate publication:             0.12 Hz
INFO  [gps] rate RTCM injection:          0.00 Hz
~~~

It sounded to me as if you were getting a hard failure. I have not seen that ever. Can you reduce your setup to just the GPS a Power module and the console? On the console issue 'gps status' every .25-.5 Seconds (I use putty right click) and see it it does start eventually.

tops4u, 2018-06-11:
Did you Update the Bootloader according to the docs? Otherwise it would identify as FMUv2

davids5, 2018-06-11:
Nope! My Bad. I just used it out of the box. I see that section....I just skipped to the pretty picture to hook it up.

davids5, 2018-06-11:
Ok. Even with the correct bootloader QCG is having issues finding the board. 
[image]

regardless installing  your bin GPS is always coming up. 

@tops4u  It sounded to me as if you were getting a hard failure. I have not seen that ever. Can you reduce your setup to just the GPS a Power module and the console? On the console issue 'gps status' every .25-.5 Seconds (I use putty right click) and see it it does start eventually. 

Is it true that no matter how long you wait your GPS never starts? 

Do you have a link to the actual GPS you are using?

tops4u, 2018-06-11:
I use a Mac. Never had issues finding the Board. 

Tried with Drotek uBlox M8N / M8Q. They work like a charm on Dropix regardless of Build/Version.

bkueng, 2018-06-12:
@davids5 the following binary fails for sure, it is f87fa9131b70735047d37bf7e06db1d613389fec, GCC 7.1.0, and I tested with USB power.
[auav-x21_default_f87fa9131b70735.zip](https://github.com/PX4/Firmware/files/2093389/auav-x21_default_f87fa9131b70735.zip)

davids5, 2018-06-12:
@bkueng - thank you! I loaded the elf and tested with only a) USB b) a debug cable and c) the GPS connected results are the same as noted above:
GPS does start after bootup

```
nsh> [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: AUAV_X21
FW git-hash: f87fa9131b70735047d37bf7e06db1d613389fec
FW version: 1.8.0 0 (17301504)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ac630e6d0e90480121c74d59a92676f0b951dce
Build datetime: Jun 12 2018 11:11:39
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.1.0
MFGUID: 363533353336510b00210031
MCU: STM32F42x, rev. 3
UID: 210031:3336510B:36353335
[hardfault_log] Fault Log is Armed
INFO  [tune_control] Publishing standard tune 1
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
nsh: rgbled_pwm: command not found
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
WARN  [hmc5883] no device on bus 1 (type: 2)
LIS3MDL_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
INFO  [lis3mdl] Poll rate set to max (80hz)
INFO  [lis3mdl] Set mag range to 4 Gauss
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 1 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
MPU9250 on SPI bus 1 at 4 (1000 KHz)
INFO  [mpu9250] accel cutoff set to 30.00 Hz
INFO  [mpu9250] gyro cutoff set to 80.00 Hz
WARN  [lis3mdl] mag cal status changed
INFO  [load_mon] stack check enabled
INFO  [px4io] default PWM output device
ERROR [px4io] already loaded
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [uavcan] sensor bridge 'gnss' init ok
INFO  [uavcan] sensor bridge 'mag' init ok
INFO  [uavcan] sensor bridge 'baro' init ok
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x_can.main.mix on /dev/uavcan/esc
WARN  [uavcan] GNSS ORB fd 12
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
[... 155 lines cut ...]
        jamming_indicator: 29
        vel_m_s: 0.000
        vel_n_m_s: 0.000
        vel_e_m_s: 0.000
        vel_d_m_s: 0.000
        cog_rad: 0.000
        timestamp_time_relative: 0
        fix_type: 0
        vel_ned_valid: 0
        satellites_used: 0
INFO  [gps] rate position:                7.22 Hz
INFO  [gps] rate velocity:                7.22 Hz
INFO  [gps] rate publication:             0.09 Hz
INFO  [gps] rate RTCM injection:          0.00 Hz
``` 

Are you connected to QGC?
Do your boards look to be the same as mine show above?
Do you have only the one cable connected to the GPS?

I am using a isolated debug console - it is an ftdi that will not back feed power to the FMU while connected and FMU is not powered. - I tested with the isolation switched off and it still will not fail. But the it worth a try on your side. Connect the FTDI after the unit is powered.

What I have noted is that after the GPS has been on for a while then I disconnect the FMU's usb and reconnect it to power cycle the GPS is ready sooner. This may be some sort of a cold start vs warm start if dT of the last fix is small and boot is quicker.  

Could that have clouded the testing?

davids5, 2018-06-12:
Hi @tops4u where do you see  12.6? 
All the PX4 files provided above have been tested. The last one was from @bkueng's zip file

tops4u, 2018-06-12:
Check your Bootlog above

`nsh> [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: AUAV_X21
FW git-hash: f87fa9131b70735047d37bf7e06db1d613389fec
FW version: 1.8.0 0 (17301504)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ac630e6d0e90480121c74d59a92676f0b951dce
*Build datetime: Jun 12 2018 11:11:39*
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.1.0
MFGUID: 363533353336510b00210031
MCU: STM32F42x, rev. 3
UID: 210031:3336510B:36353335`

tops4u, 2018-06-12:
@davids5 

Just created a Log from the AUAV-X2 (FMUv2) which suffers the same Problem. 

Picture of the Setup (most simple, a uBlox M8Q and the Console just GND, TX, RX):
[image]

Log after issuing REBOOT:
```
reboot
WARN  [platforms__common] Reboot NOW.
WARN  [ecl/EKF] EKF stopping navigation
WARN  [driv
FMUv2 ver 0x8 : Rev 0 V2
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1

[boot] Fault Log is Armed

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4FMU_V2
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: f87fa9131b70735047d37bf7e06db1d613389fec
FW version: 1.8.0 80 (17301632)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ac630e6d0e90480121c74d59a92676f0b951dce
Build datetime: Jun  6 2018 21:04:31
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 393231353333470f003e0030
MCU: STM32F42x, rev. Y

WARNING   WARNING   WARNING!
Revision Y has a silicon errata:
This device can only utilize a maximum of 1MB flash safely!
https://pixhawk.org/help/errata

UID: 3E0030:3333470F:39323135 
nsh: mount: mount failed: No such device
INFO  [tone_alarm] missing command, try 'start', status, 'stop'
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [tune_control] Publishing standard tune 1
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #2
nsh: blinkm: command not found
nsh: rgbled_pwm: command not found
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
MS5611_SPI on SPI bus 4 at 3 (20000 KHz)
WARN  [ms5611] no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
nsh: bst: command not found
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 4 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
WARN  [mpu6000] no device on bus #5 (SPI4)
WARN  [mpu9250] probe failed! 104
WARN  [mpu9250] no device on bus 3
WARN  [mpu9250] probe failed! 0
WARN  [mpu9250] no device on bus 5
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [sf1xx] driver start failed
ERROR [param] Parameter SENS_EN_MB12XX not found
ERROR [param] Parameter SENS_EN_LEDDAR1 not found
INFO  [load_mon] stack check enabled
ERROR [param] Parameter UAVCAN_ENABLE not found
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Onboard, data rate: 5000 B/s on /dev/ttyS2 @ 57600B
WARN  [mavlink] stream SCALED_IMU2 not found
WARN  [mavlink] stream SCALED_IMU3 not found
ERROR [param] Parameter UAVCAN_ENABLE not found
px4flow [217:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Mixer: /etc/mixers/hexa_x.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 
INFO  [tone_alarm] missing command, try 'start', status, 'stop'
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: /fs/microsd/log

NuttShell (NSH)
nsh> gINFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
ps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 19200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 57600, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: UBX
INFO  [gps] port: /dev/ttyS3, baudrate: 9600, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: ASHTECH
INFO  [gps] port: /dev/ttyS3, baudrate: 115200, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> gps status
INFO  [gps] Main GPS
INFO  [gps] protocol: MTK
INFO  [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO  [gps] sat info: disabled
nsh> 
nsh> 
```

davids5, 2018-06-12:
>Check your Bootlog above

>>nsh> [boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: AUAV_X21
FW git-hash: f87fa91
FW version: 1.8.0 0 (17301504)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ac630e6d0e90480121c74d59a92676f0b951dce
Build datetime: Jun 12 2018 11:11:39
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.1.0
MFGUID: 363533353336510b00210031
MCU: STM32F42x, rev. 3
UID: 210031:3336510B:36353335

The files @bkueng sent have that time/date. 
[image]

I am not seeing the problem you are pointing out. What am I missing?

tops4u, 2018-06-12:
@davids5 : First Log ist with Master from last week -> GPS fails after reboot. Second Log is with 1.7.0 from December where GPS starts on the spot after booting. Same Setup, same HW, etc just flashed older FW.

davids5, 2018-06-12:
@tops4u - I understand the content of https://github.com/PX4/Firmware/issues/9461#issuecomment-396704387. The June 6 vs June 12 part (6.12) was where you had me confused from https://github.com/PX4/Firmware/issues/9461#issuecomment-396688254

The https://github.com/PX4/Firmware/issues/9461#issuecomment-396647008 test results was with FW @bkueng sent of his build that he has failing on the X2.1 HW you sent him. Are you thinking he sent me the wrong files?

I placed his FW on the HW I have a picture above of.  That works.

You are also seeing the issue on your X2 HW with a different GPS and FMU than I am testing with.

Things that may be different:

1. Power backfeeding from FTDI
  Do you disconnect the FMU USB and FTDI (i.e. remove all power from the system) when change firmware and retest?

2. GPS is not the same as I am testing with.

If I can not replicated it with my HW, I will most likely need the HW @bkueng has to track this down.

tops4u, 2018-06-13:
@davids5 - Regarding the Questions:

1. Yes FTDI had 5V disconnected and the USB Connection as well as the USB to FTDI has been cut between tests. I doubt this is a GPS Init Problem. As mentioned above I have issued the GPS STATUS manually by typing it each time. This took around 30 - 60 secs to do in the first log and still the GPS not detected, whereas with the old FW I just typed it immediately after boot, so around 3-4 secs later GPS was ready.
2. GPS ist a uBlox m8q but the same also happens with a m8n Unit. Both are from Drotek (so I assume they are genuine uBlox). I have never had any issues with those units prior to Release 1.7.3 and later. 

I have not yet tested with Beat's FW but I can do this today evening. I have no spare mRo Unit so I tested with the older AUAV X2, the only mRo I have left is in a Hexa. Let me test this later today.

tops4u, 2018-06-13:
@bkueng I am not sure, build infrastructure is relevant. When I build master on my Mac with GCC 5.x it does work as well, only the Server Buildchain fails for me. The Toolchain you used is not identical to the one on the Server (_Toolchain: GNU GCC, 7.1.0_ vs. _Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]_)

Did your Build fail on your FMU?

bkueng, 2018-06-13:
> Did your Build fail on your FMU?

Yes

tops4u, 2018-06-13:
@bkueng Ok, I'll check with your build on my mRo Unit later today.

davids5, 2018-06-13:
@tops4u, @bkueng - the parameters setting are the reason I could not reproduce this. THANK YOU @bkueng. 

The issue was the camera trigger code. The Fix is here https://github.com/PX4/Firmware/pull/9665

davids5, 2018-06-13:
@tops4u, @bkueng - Thank you for your help!

All of this make sense. All the weirdness: What compiler, how it was built would have changed the data indexed by the -1 on the gpio array. Why the reopen worked. 

The data at gpio[-1] passed to the px4_arch_configgpio would have affected some IO on the CPU. Yikes!!!. In this specific case UART4 TX PA0 was overwritten to an Analog input by the camera trigger code bug.

tops4u, 2018-06-13:
Of course! I have configured the Camera on the AUAV-X2 and the mRo Unit but not on the Dropix B-)

Good finding! Thanks a lot @davids5 & @bkueng

bkueng, 2018-06-14:
Cool! Nice work everyone!
I can confirm that #9665 fixes it.

closed 2018-06-14 by LorenzMeier