PX4/PX4-Autopilot #9463: GPS here+ plus rover is not getting detected

siddyyagno3, 2018-05-14:
I have a Here plus rtk gps. The rovers gps module, when connected to pixhawk 2 cube and loaded with px4 stable is not locking to gps. I tried leaving it out for half hour. It is blinking blue light and GPS count is 0. But if I flash it with ArduPilot it is locking in seconds. Am I missing some parameter change. 
NSH command GPS status
NuttShell (NSH)
nsh> gps status
INFO [gps] Main GPS
INFO [gps] protocol: MTK
INFO [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO [gps] sat info: disabled, noise: 0, jamming detected: NO
nsh> gps status
INFO [gps] Main GPS
INFO [gps] protocol: MTK
INFO [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO [gps] sat info: disabled, noise: 0, jamming detected: NO
nsh> gps status
INFO [gps] Main GPS
INFO [gps] protocol: UBX
INFO [gps] port: /dev/ttyS3, baudrate: 38400, status: NOT OK
INFO [gps] sat info: disabled, noise: 0, jamming detected: NO
Also, I wanted to know that I can set the RTK survey in accuracy to lowest 0.5. So is this parameter is congruous to the overall accuracy of the setup. Or leaving the setup for more time will eventually increase the accuracy to 10cm level.

Thanks in advance. 
Regards,
Sidd

siddyyagno3, 2018-05-14:
GPS 1 port on hardware pixhawk2 cube

LorenzMeier, 2018-05-15:
@bkueng Anything that comes to mind?

mhkabir, 2018-05-15:
Potentially related : https://github.com/PX4/Firmware/issues/9461#issuecomment-388973962

tops4u, 2018-05-15:
@siddyyagno3 try >gps stop / >gps start if needed 2-3x and check if your GPS is then found. If so, we may have the same Problem (#9461). Also it may work prior to 1.7.4.

siddyyagno3, 2018-05-17:
I tried rolling back to 1.70 didn't help.

siddyyagno3, 2018-05-17:
@bkueng @mhkabir @LorenzMeier 
Can I rule out that issue is due to configuration as in guide "The PX4 GPS stack automatically sets up the u-blox M8P modules to send and receive the correct messages over the UART or USB, depending on where the module is connected (to QGroundControl or the autopilot)."
When the rover module is connected using usb it is detected as rtk.

bkueng, 2018-05-18:
> So, It might not be because i might have changed some config in ublox center.

Most likely not, but you can try resetting it to factory defaults.

bkueng, 2018-05-24:
I got my hands on a Here+, which is attached to a Pixhawk Pro. It works fine when flashing stable.

> But When I tried nsh> gps status protocol is showing as Ashtech not UBX

It means it's using the NMEA protocol instead of the u-blox protocol. This should not happen if you use the default config.

> gps error failed to instantiate object

The board is probably out of memory. A reboot will help.

siddyyagno3, 2018-05-24:
Found the problem and Murphy was involved so the issue was with the Pixhawk2 cube carrier board's faulty gps port, some internal wiring and probably inadequate soldering. I found a spare DF13 connector and connected it to nice old pixhawk and everything was working.  I was banging my head on configuration or gps hardware but both were fine. Quality should be ramped up on Pixhawk2 cube carrier boards. Still could not get the 3d RTK Fix but 3D rtk float made me happy. Will be changing the setup to improve signal strength. 
Thanks for the help and sorry for inconvenience.

closed 2018-05-24 by siddyyagno3

bkueng, 2018-05-24:
Glad to see that you found it, thanks for letting us know.