PX4/PX4-Autopilot #9466: "make posix_sitl_default gazebo" error

alexcherpi, 2018-05-15:
Hi,

I cloned the PX4 Firmware v1.7.3.
I am able to launch the jmavsim simulator with the command "make posix_sitl_default jmavsim" and it works fine.
If I run gazebo it's working fine on its own.
My gazebo version is Gazebo9 and I m working on ubuntu 16.04LTS
But when I run the command "make posix_sitl_default gazebo" I get the following error log:

[1/1] Re-running CMake...
-- Build Type: RelWithDebInfo
-- PX4 VERSION: v1.7.3
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu116.04.9) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu116.04.9) 5.4.0 20160609
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/build/posix_sitl_default
[7/18] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
-- thread
-- signals
-- system
-- filesystem
-- program_options
-- regex
-- iostreams
-- date_time
-- chrono
-- atomic
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "2.3.0")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
-- Found libzmq , version 4.1.4
-- Checking for module 'uuid'
-- Found uuid, version 2.27.0
-- Checking for module 'tinyxml2'
-- Found tinyxml2, version 2.2.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
-- Found gts, version 0.7.6
-- Checking for module 'libswscale'
-- Found libswscale, version 3.1.101
-- Checking for module 'libavdevice >= 56.4.100'
-- Found libavdevice , version 56.4.100
-- Checking for module 'libavformat'
-- Found libavformat, version 56.40.101
-- Checking for module 'libavcodec'
-- Found libavcodec, version 56.60.100
-- Checking for module 'libavutil'
[... 174 lines cut ...]
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:247:3: error: ‘math’ has not been declared
math::Quaternion C_W_I = T_W_I.rot;
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:251:22: error: ‘C_W_I’ was not declared in this scope
orientation->set_x(C_W_I.x);
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:274:3: error: ‘math’ has not been declared
math::Vector3 acceleration_I = link_->GetRelativeLinearAccel() - C_W_I.RotateVectorReverse(gravity_W_);
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:277:3: error: ‘math’ has not been declared
math::Vector3 angular_vel_I = link_->GetRelativeAngularVel();
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:279:41: error: ‘acceleration_I’ was not declared in this scope
Eigen::Vector3d linear_acceleration_I(acceleration_I.x,
^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:282:38: error: ‘angular_vel_I’ was not declared in this scope
Eigen::Vector3d angular_velocity_I(angular_vel_I.x,
^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

Do you know what's wrong ?

alexcherpi, 2018-05-17:
I solved my issue.
I removed gazebo using:
sudo apt-get remove '.*gazebo.*' '.*sdformat.*' '.*ignition-math.*' '.*ignition-msgs.*' '.*ignition-transport.*'
Then I installed ros-kinetic-desktop-full using:
sudo apt-get install ros-kinetic-desktop-full
Then I ran the bash script from px4 dev guide:
ubuntu_sim_ros_gazebo.sh 

And then I was able to launch the px4-gazebo simulation:
make posix_sitl_ekf2 gazebo
Even if it s not the the ROS version. Now my gazebo version is Gazebo version 7.12.0 (before version 9) which maybe solved my issue.
And I am also able to launch px4-ros-gazebo simulation:
roslaunch px4 posix_sitl.launch

closed 2018-05-17 by alexcherpi

alexcherpi, 2018-05-18:
@hamishwillee Yeah the Gazebo version 9 comes with the script [ubuntu_sim.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim.sh) you mentioned. Gazebo 7 comes with the ros version [ubuntu_sim_ros_gazebo.sh](https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_ros_gazebo.sh). The issue with me (shown in the my previous comment in this issue) was with Gazebo 9. But maybe I an isolated case.

hamishwillee, 2018-05-21:
@alexcherpi Thank you. I suspect that the fixes that enabled support for Gazebo 9 came in after v1.7.3. I tested them on master.