PX4/PX4-Autopilot #9472: Mission feasibility checks: add tolerance for landing pattern check

tstastny, 2018-05-16:
Currently, uploading a landing pattern calculated on QGC can sometimes lead to floating point errors (likely due to global/local conversions back and forth) causing the criteria for the landing mission to be violated and the mission thusly rejected. I think adding a small tolerance here: [mission_feasibility_checker.cpp#L414](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_feasibility_checker.cpp#L414), would be sufficient. I've seen that the difference can be somewhere in the range of 0 to 1 m. Often around 0.5m.

This also assumes, however, that either the user is sending appropriately placed waypoints, or that the corresponding addition in QGC is also made: see mavlink/qgroundcontrol#6457

I could PR this quite quickly, (just adding one number), but open to thoughts / ideas. @dagar

philipoe, 2018-05-16:
I'd actually label this as a "bug" and not an "enhancement", because last time (testing autoland with PX4 + QGC) it took us around 5 minutes to define the proper landing approach solely due to this bug, which i don't is acceptable for day-to-day operations.

LorenzMeier, 2018-05-16:
Adding the tolerance on the Firmware side is fine. The root cause is that QGC should be using MISSION_INT and not using it results in floating point errors in the order of 2 meters. @DonLakeFlyer Is this being used consistently already?

DonLakeFlyer, 2018-05-16:
> The root cause is that QGC should be using MISSION_INT and not using it results in floating point errors in the order of 2 meters.

If the vehicle reports it supports MISSION_INT then QGC sends in MISSION_INT format.

philipoe, 2018-06-20:
@tstastny Is this still an open issue or was this fixed somehow in the meantime?

tstastny, 2018-06-20:
> @tstastny Is this still an open issue or was this fixed somehow in the meantime?

I would say that it still makes sense to add a small tolerance to the IF condition. But the primary issue is on the QGC side -- calculating the FW landing pattern considering flare params (the calculations should match on QGC and Firmware side, or else it will of course give errors).

closed 2019-01-23 by tstastny