PX4/PX4-Autopilot #9477: Airframes Reference Rover is not shown in QGC

wkyoun, 2018-05-17:
@TSC21 

Airframes Reference Rover is not shown in QGC

I have changed several version of QGC, but airframes Reference Rover is not shown in QGC

TSC21, 2018-05-17:
@wkyoun the generic rover airframe is called `Generic Ground Vehicle`, and its config file is `50000_generic_ground_vehicle` (you can check the airframe reference here: https://dev.px4.io/en/airframes/airframe_reference.html). You should be able to choose it from the Airframe setup (https://docs.qgroundcontrol.com/en/SetupView/Airframe.html). As you can see in latest master for QGC, Rover metadata exists: https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/AirframeFactMetaData.xml#L563. Same for [Stable_V3.3](https://github.com/mavlink/qgroundcontrol/blob/Stable_V3.3/src/AutoPilotPlugins/PX4/AirframeFactMetaData.xml#L563).
Did you tried latest stable release? Can you show me what Airframe references appear? I think if this is a problem, it's not related with PX4 but rather with QGC.

hamishwillee, 2018-06-19:
@El7ias As per https://github.com/PX4/Firmware/issues/9477#issuecomment-389834678 , did you check that the vehicle named "Generic Ground Vehicle" did not exist?

If not, @TSC21 is arguing that this is a QGC issue, and should be posted there.

closed 2019-02-06 by stale[bot]