PX4/PX4-Autopilot #9478: Navigation Based on Local Coordinates

mrivi, 2018-05-17:
With the growing importance of vision based navigation, our current navigation pipeline is a strong limitation since everything is handled in global coordinates.
Simple features like auto-takeoff, hold are not possible because trying to support them in both local and global coordinates is impossible with the current spaghetti structure of the navigator and commander modules.

@ChristophTobler  and I tried several approaches and we got blocked every time. Here a recap:
- **reference altitude**: if the waypoint is created in local frame as soon as global coordinates are available the reference altitude is initialized. The check for global position pass and therefore the reference altitude is subtracted from the waypoint because there is no notion that that waypoint was in local coordinates.
- **mission item**: we have tried to add local takeoff and loiter to the navigator along side global support. The mission item has to support local coordinates otherwise the takeoff cannot terminate since the mission item has to be reached to switch to loiter
- **relative_altitude flag**: the mission item relative altitude is present but not supported
- **loiter**: requires global coordinates as it is now implemented

The current architecture limitations are difficult to overcome because they require a large amount of changes even to have small features like a takeoff in local coordinates. 

FYI - @dagar

ChristophTobler, 2018-05-17:
@Stifael and I already tried getting rid of the global dependency here https://github.com/PX4/Firmware/pull/7432 but that got canceled/forgotten because of the lacking support...
I really think we need this soon otherwise e.g. vision will never be integrated in a sustainable way

Side note: I vote for a commander/navigator rewrite! It's just a huge pain (or even impossible) to fix it in small portions because of the spaghetti code

LorenzMeier, 2018-05-17:
How flexible are you next week for a vision focused meeting? 1-4 pm on Wednesday?

TSC21, 2018-05-17:
I will be interested on taking part on that effort too. But for the call, those times are only possible to me on Fridays.

px4Li, 2018-12-30:
Hi all, if I want to use vision position in mission mode, so I have to convert local position x and y to lat and lon. Is that correct? If it is true, should I use mavlink sent the lat and lon as fake gps to mission mode?

TSC21, 2018-12-30:
@px4Li as you can read above, you cannot currently navigate on mission mode if you don't have a global position, unless you use the LPE + Attitude Estimator Q combination, since LPE actually allows to create a virtual global origin by setting the `LPE_FAKE_ORIGIN` parameter. Though, we are not currently actively supporting LPE.

hardeman, 2019-04-05:
Hello. Is mission mode based on local coordinates already available? Or will this be available with the 1.10.0 release?

TSC21, 2019-04-05:
Not available yet.

mrivi, 2019-10-02:
still relevant

TzionCastillo1, 2025-07-13:
Any update on when this might be added?

dakejahl, 2025-07-13:
> Any update on when this might be added?

No one is working on it. Navigator needs a rewrite in order to support it. See https://github.com/PX4/PX4-Autopilot/pull/24335#issuecomment-2660156287