PX4/PX4-Autopilot #9479: LPE- mocap fused but local_pose not synchronize

glennliu, 2018-05-17:
Hi 

I'm using PX4+Odroid+mavros with Optitrack system. Mocap data is published to PX4 through onboard Odroid Xu4's UART. In my first test, I control the drone manually with mocap data streamed to PX4. local_pose and mocap_pose plot are showed as below,
 
**mocap_x Vs local_pose_x**
[image]

**mocap_y Vs local_pose_y**
[image]

**mocap_z Vs local_pose_z**
[image]

 As it shows, local_pose cannot follow mocap_pose data, which means it doesn't receive good estimation. I tried decrease LPE_VIC_P, LPE_VIS_Z to minimum value. LPE_PN_V is increased to maximum value. But it doesn't improve much. However, when drone is static on the ground. local_pose can follow mocap_pose nicely. 
So, my question is how to receive good estimation in flying? 

The main specification are:
Firmware: PX4 1.7.2
Flight controller: MindRacer
Full log attached:
[log_34_2018-5-16-13-13-50.zip](https://github.com/PX4/Firmware/files/2013594/log_34_2018-5-16-13-13-50.zip)

Thanks in advance

glennliu, 2018-05-18:
Tried to increase mocap steram rate to 60hz. Results didn't change much.
[image]

glennliu, 2018-05-18:
I put the drone on the ground static at different position. Local_pose can follow mocap_pose well. It just take a bit long response time.
[image]

glennliu, 2018-05-20:
Turns out its an wifi issue. Cause I used 5G wifi on Odroid to transmit mocap data.
When I tried ethernet to transmit mocap, results looks correct

[image]

jgoppert, 2018-11-13:
Seems resolved.

closed 2018-11-13 by jgoppert