PX4/PX4-Autopilot #9495: Console output on Mac OS

LorenzMeier, 2018-05-21:
@MaEtUgR M
FYI

> Makefile colors do not work e.g. \033[0;34mFormatting with astyle \033[m
> Certain terminal combinations like cygwin - mintty - xterm - bash do not escape colors by default.
> Solution: Force interpretation of backslash escapes with the parameter -e of echo.

This change introduce a new issue in MacOS console:
```
bash-3.2$ make format
-e \033[0;94mFormatting with astyle \033[m
```

PR that introduces this: https://github.com/PX4/Firmware/pull/9441#issuecomment-390569527

LorenzMeier, 2018-05-28:
Fixed!

closed 2018-05-28 by LorenzMeier

MaEtUgR, 2018-05-29:
@LorenzMeier ok, thanks. Would be interesting how, I don't see any change. Still glad the problem doesn't seem to persist.