PX4/PX4-Autopilot #9500: Bug Report: Multi-Vehicle Interference in the storage and execution of the mission

Ahrovan, 2018-05-22:
This code just specifies each uav's move to a specific location. when I use two uav's or more (our main code with 5 uav's) after push mission and send "arm", all uav's moved to just one location not to the location specified in their mission.
[image] - We used upstream version of PX4, Mavros and follow the document dev.px4.io/en/simulation/multi-vehicle-simulation.html

mavros code:
```
...
  point.x_lat = 47.398232;
  point.y_long = 8.545628;
  record_0.waypoints.push_back(point);
  ROS_INFO("Record Point uav_0:[%f,%f]", point.x_lat, point.y_long);
  wp_list_0 = transformWaypoints(record_0, 3, 8);
  clearWaypointList("uav1/mavros/mission/clear");
  pushWaypointList("uav1/mavros/mission/push", wp_list_0);

  point.x_lat = 47.398494;
  point.y_long = 8.544589;
  record_1.waypoints.push_back(point);
  ROS_INFO("Record Point uav_1:[%f,%f]", point.x_lat, point.y_long);
  wp_list_1 = transformWaypoints(record_1, 3, 10);
  clearWaypointList("uav2/mavros/mission/clear");
  pushWaypointList("uav2/mavros/mission/push", wp_list_1);

  point.x_lat = 47.397545;
  point.y_long = 8.544957;
  record_2.waypoints.push_back(point);
  ROS_INFO("Record Point uav_2:[%f,%f]", point.x_lat, point.y_long);
  wp_list_2 = transformWaypoints(record_2, 3, 12);
  clearWaypointList("uav3/mavros/mission/clear");
  pushWaypointList("uav3/mavros/mission/push", wp_list_2);
...
```
 - Flight log 
https://ufile.io/u4njg
```
roslaunch px4 multi_uav_mavros_sitl.launch
... logging to /home/ahrovan/.ros/log/0512a88c-5d8f-11e8-83f2-000c298cb19f/roslaunch-ahrovan-92992.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ahrovan:42757/

SUMMARY
========

CLEAR PARAMETERS
 * /uav1/mavros/
 * /uav2/mavros/
 * /uav3/mavros/
 * /uav4/mavros/
 * /uav5/mavros/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /uav1/mavros/cmd/use_comp_id_system_control: False
...
 * /uav1/mavros/vision_speed/twist_cov: True
 * /uav1/rotors_description: <?xml version="1....
 * /uav2/mavros/cmd/use_comp_id_system_control: False
...
 * /uav2/mavros/vision_speed/twist_cov: True
 * /uav2/rotors_description: <?xml version="1....
 * /uav3/mavros/cmd/use_comp_id_system_control: False
...
 * /uav3/mavros/vision_speed/twist_cov: True
 * /uav3/rotors_description: <?xml version="1....
 * /uav4/mavros/cmd/use_comp_id_system_control: False
...
 * /uav4/mavros/vision_speed/listen_twist: True
[... 735 lines cut ...]
INFO  [Unknown] EKF GPS checks passed (WGS-84 origin set)
INFO  [Unknown] EKF commencing GPS fusion
INFO  [Unknown] EKF GPS checks passed (WGS-84 origin set)
INFO  [Unknown] EKF commencing GPS fusion
[ INFO] [1526972963.467123850, 11.802000000]: HP: requesting home position
INFO  [Unknown] EKF GPS checks passed (WGS-84 origin set)
INFO  [Unknown] EKF commencing GPS fusion
[ INFO] [1526972963.722932559, 12.056000000]: HP: requesting home position
INFO  [Unknown] EKF GPS checks passed (WGS-84 origin set)
[ INFO] [1526972963.992589162, 12.320000000]: HP: requesting home position
INFO  [Unknown] EKF commencing GPS fusion
[ INFO] [1526972964.197380253, 12.526000000]: HP: requesting home position
[ INFO] [1526972964.503919217, 12.830000000]: HP: requesting home position
[ WARN] [1526972965.840929184, 14.154000000]: TM: Clock skew detected (-0.010992632 s). Hard syncing clocks.
[ WARN] [1526972966.042906797, 14.356000000]: TM: Clock skew detected (-0.013094409 s). Hard syncing clocks.
[ WARN] [1526972966.094060910, 14.406000000]: TM: Clock skew detected (-0.010440717 s). Hard syncing clocks.
[ WARN] [1526972966.339976915, 14.652000000]: TM: Clock skew detected (-0.011993698 s). Hard syncing clocks.
[ INFO] [1526972968.496002709, 16.802000000]: WP: mission received
[ WARN] [1526972968.639945004, 16.944000000]: TM: Clock skew detected (-0.015382450 s). Hard syncing clocks.
[ INFO] [1526972968.753222967, 17.056000000]: WP: mission received
[ INFO] [1526972969.019065733, 17.320000000]: WP: mission received
[ INFO] [1526972969.224659405, 17.526000000]: WP: mission received
[ INFO] [1526972969.528948598, 17.830000000]: WP: mission received
```
this issues following "Interference in /.ros/rootfs #206"

lamping7, 2018-05-22:
This is not a sitl_gazebo issue. It's a firmware issue.

It's already documented: https://github.com/PX4/Firmware/issues/9046

If you follow that issue, you'll see I hacked together a quick fix.

closed 2018-05-22 by lamping7

Ahrovan, 2018-05-23:
Thank you, the problem was resolved