PX4/PX4-Autopilot #9502: "make posix_sitl_default gazebo" with error

xup682u03, 2018-05-22:
I'm trying to use mavros to control PX4 in gazebo under ubuntu mate.
When I try to run "make posix_sitl_default gazebo", it failed everytime.
I google a lot and rebuild the whole system once but it didn't work.

The following is my error message:

[image]

[image]
 

It confused me for a long while.
Thanks in advance.

hamishwillee, 2018-05-22:
Are you on a Ubuntu machine or within a virtual environment?

Is this current master Firmware?

Did you run one of the scripts here https://dev.px4.io/en/setup/dev_env_linux.html to set up your developer environment?

LorenzMeier, 2018-05-22:
I'm closing this since the system `gz` program segfaulted. This has nothing to do with PX4 or the toolchain but is a Linux installation issue. Please install the latest Linux updates or updated version.

xup682u03, 2018-05-22:
@hamishwillee 
Oops, forgot to mention.
I'm using ubuntu on RPI3. 
I will set up the developer environment and report the result. thanks.

xup682u03, 2018-05-22:
@LorenzMeier 
Hi, thanks for reply.
Do you mean `sudo apt-get update`?

xup682u03, 2018-05-23:
@hamishwillee 
Hi, this is the latest situation after `source ubuntu_sim_ros_gazebo.sh`.

[image]

Error in `gz': double free or corruption (out):  0x014ec748 ***
[image]

ninja: build stopped" subcommand failed. 
Makefile:147 recipe for target 'posix_sitl_default' failed
[image]

TSC21, 2018-10-07:
@xup682u03 I don't think it's reasonable to actually run Gazebo in a RPi at all. I do advise you to use it on a standard x86 machine, as you will face obvious problems of performance to run this on a RPi. Closing this now.

closed 2018-10-07 by TSC21