PX4/PX4-Autopilot #9505: 1.8.0beta does not "work" with QGC latest

jstrebel, 2018-05-22:
Hi,
I tried v1.8.0 (beta) f09f3058 and QGC 19eed083d 2018-05-21 18:23. With this setup, I get 7 times the error message "Critical: IGN MISSION_ITEM: No transfer"
QGC runs on OSX 10.13.5, Telemtrie Link is (433Mhz Telemetrie set) Serial 56kbps
[schafisheim18beta.plan.zip](https://github.com/PX4/Firmware/files/2027768/schafisheim18beta.plan.zip)

HW is PXFMU_V2 (V2)
when I upload the mission attached. It's impossible to ARM PIXhawk.
If you need more details let me know. Jakob

jstrebel, 2018-05-22:
here is the QGC log:
[qgc_log_latest.txt](https://github.com/PX4/Firmware/files/2027809/qgc_log_latest.txt)

Antiheavy, 2018-05-23:
We’ve also been getting these IGN errors using some old 3DR modems with 1.8.0beta

ryanjAA, 2018-05-23:
Been getting the same ign messages on Rfd900 on one end and 3DR units (both actual 3DR ones and ones from Asia). Still uploads though and flies fine. Probably 10-15 flights with this.

dagar, 2018-05-23:
@julianoes any ideas here?

LorenzMeier, 2018-05-23:
Which version of QGC are you running? Is any type of OSD connected?

jstrebel, 2018-05-23:
@Lorenz I run daily build QGC 19eed083d 2018-05-21 18:23. What does OSD mean, sorry for the ignorance.

hamishwillee, 2018-05-23:
OSD is "On screen display". A board that pulls telemetry data from your flight controller and over-lays it on your First Person View monitor.

jstrebel, 2018-05-23:
Thank you. QGC runs on OSX. All Telemetry Info is displayed on this Screen.

ryanjAA, 2018-05-23:
Daily versions from last Friday through the several weeks prior. Apologizes I don’t have the exact versions but I can track them down. It happened so frequently we just finally ignored it. OSD I will check if it was turned on for telem2. Definitely was not using one but it may have been enabled. Spoke to @Dagar briefly about this. Could it be log streaming in qgc since that is kicking out the not enough bandwidth and therefor busy ign transfer message?

jstrebel, 2018-05-23:
Log streaming was not enabled in QGC. It was impossible to ARM the pixhawk. However, with QGC 3.2.2 I was able to ARM the pixhawk and fly.

bkueng, 2018-05-28:
I can reproduce this with QGC master and an older version (f1e2638cad24b07b1b and 1e856cc). It means that the vehicle received a mission item, but it was not in upload mode (it did not get a MAVLINK_MSG_ID_MISSION_COUNT before).

Here's the transfer output:
```
DEBUG 133524503  WPM: Send MISSION_REQUEST_INT seq 10 to ID 255
DEBUG 133713551  WPM: MISSION_ITEM ERROR: seq 9 was not the expected 10
DEBUG 133714000  WPM: Send MISSION_REQUEST_INT seq 10 to ID 255
DEBUG 133724399  WPM: MISSION_ITEM ERROR: seq 9 was not the expected 10
DEBUG 133724770  WPM: Send MISSION_REQUEST_INT seq 10 to ID 255
DEBUG 133743929  WPM: MISSION_ITEM seq 10 received
DEBUG 133744258  WPM: Send MISSION_REQUEST_INT seq 11 to ID 255
DEBUG 133942536  WPM: MISSION_ITEM ERROR: seq 10 was not the expected 11
DEBUG 133942854  WPM: Send MISSION_REQUEST_INT seq 11 to ID 255
DEBUG 133953537  WPM: MISSION_ITEM ERROR: seq 10 was not the expected 11
DEBUG 133953935  WPM: Send MISSION_REQUEST_INT seq 11 to ID 255
DEBUG 133966310  WPM: MISSION_ITEM seq 11 received
DEBUG 133966559  WPM: MISSION_ITEM got all 12 items, current_seq=0, changing state to MAVLINK_WPM_STATE_IDLE
DEBUG 133971091  WPM: Send MISSIDEBUG 133971375  WPM: New missioON_ACK type 0 to ID 255
DEBUG 133975380  WPM: New mission detected (possibly over differDEBUG 133979372  WPM: New mission detected (possibly over different Mavlink instance) Updating
n detected (possibly over different Mavlink instance) Updating
DEBUG 134010609  WPM: Send MISSIent Mavlink instance) Updating
DEBUG 134021891  WPM: Send MISSION_COUNT 12 to ID 0, mission typDEBUG 134033076  WPM: Send MISSION_COUNT 12 to ID 0, mission type=0
ON_COUNT 12 to ID 0, mission type=0
e=0
DEBUG 134178415  WPM: MISSION_ITEM ERROR: no transfer
WARN  [mavlink] IGN MISSION_ITEM: No transfer
ERROR [mavlink] IGN MISSION_ITEM: No transfer
DEBUG 134179392  WPM: Send MISSION_ACK type 1 to ID 255
DEBUG 134185112  WPM: MISSION_ITEM ERROR: no transfer
WARN  [mavlink] IGN MISSION_ITEM: No transfer
ERROR [mavlink] IGN MISSION_ITEM: No transfer
DEBUG 134189134  WPM: Send MISSION_ACK type 1 to ID 255
DEBUG 134203399  WPM: MISSION_COUNT 0, clearing waypoints list and staying in state MAVLINK_WPM_STATE_IDLE
DEBUG 134222777  WPM: Send MISSION_ACK type 0 to ID 255
DEBUG 134377383  WPM: MISSION_COUNT 0, clearing waypoints list and staying in state MAVLINK_WPM_STATE_IDLE
DEBUG 134382164  WPM: Send MISSION_ACK type 0 to ID 255
```

The received mission item is a duplicate of the previously received one (same `seq`). The mission manager already received the last item and finished the transfer but QGC did not receive/handle the last ack and sent the item again. @DonLakeFlyer this happens consistently, so might be worth looking into why the last item is sent again from QGC.
PR on the firmware side: #9547

While testing, I came across another issue: https://github.com/mavlink/qgroundcontrol/issues/6519

closed 2018-05-28 by LorenzMeier

jstrebel, 2018-06-06:
Thank you. Never had this issue again. Jakob

PR #9547 (2018-05-28): mavlink mission: send an ack on duplicated last uploaded mission item
In the case a GCS would re-send the last mission item, which already successfully was received (but the GCS did not receive/handle the ack), the mission manager would send the message "IGN MISSION_ITEM: No transfer".
This PR fixes that.

Fixes #9505