PX4/PX4-Autopilot #9523: Logic bug in sensors main program

zarathustr, 2018-05-24:
Recently we find out that the sensors module will switch to worse sensor once the better sensor reaches a TOUT error. The program will not turn to use the better sensor when the connection is recovered. Actually at this time the sensor sampling timer loop has not been terminated. So the current state should be changed to use the better sensor as soon as it is again detected as good.

dagar, 2018-05-24:
This was an intentional to prevent sensors from ping ponging between healthy or not. Once a sensor is found to be unhealthy for any reason it's demoted for the entire session.

In the case of an external (preferred) sensor perhaps you add additional logic to restore the original priority after a relatively long probation period?

zarathustr, 2018-05-26:
@dagar There is a real exceptional case. Let us suppose we have two IMUs mounting on the vehicle. While IMU1 is much better than IMU2 and less likely to break in real use. But unfortunately, bad thing happens, the IMU1 triggered an error randomly just ahead of IMU2. Then the system will automatically switch to IMU2 even if the IMU1 is recovered shortly. So I am going to enhance this point.

dakejahl, 2018-06-16:
Has there been any work to support this? We have 2 IMUs, one isolated and one mounted directly to the pcb. Occasionally we see a fail to disarm on land (lots of accel/gyro noise) and have been unable to explain it. I wonder if it is this exact scenario.

LorenzMeier, 2018-06-16:
@zarathustr The timeout margins are generous, so that should not happen in normal operation anyway.

@dakejahl Could you please provide a logfile?

zarathustr, 2018-06-17:
@LorenzMeier Hi Lorenz, That is true. Please see our pull request for the AMOV-IMU. We have made sure that 3 AMOV-IMUs can be sampled synchronously via the pixhawk autopilot for a whole week continuously (7x24 hours) without a single fault. This has been confirmed both by the perf and sensors status. No failsafe events has been triggered. We also have done the repeatability tests on various autopilots and they all run at the same status.

dakejahl, 2018-06-20:
@LorenzMeier Sorry, I think this is a hardware issue on our end with a loose connector.

zarathustr, 2018-06-21:
@dakejahl Not always. But for you maybe it was the connection problem. In most time, the error is triggered by the long wiring.