PX4/PX4-Autopilot #9532: fmu_v5 NSH some commands force an instant reboot

ryanjAA, 2018-05-26:
Certain NSH commands via QGC on Master and Beta send it to a instant reboot.

For instance ver all works fine, tests all, etc but any of the airspeed commands send it to a reboot. Plugged in directly via usb only and only one airspeed (sdp3x) at a time plugged in (to i2c a). Even with nothing plugged in it does the same thing reboots.

dagar, 2018-05-26:
@DanielePettenuzzo any ideas? All currently supported differential pressure sensors (airspeed) are i2c.

ryanjAA, 2018-05-26:
I am wrong, that command doent reboot it but it also wont let the sdp3x start.

DanielePettenuzzo, 2018-05-26:
@ryanjAA if you connect the sensor to i2c-a and start the driver what message do you get on the console?
sdp3x_airspeed start -b 4

ryanjAA, 2018-05-26:
Two firmware flashs and all set. Maybe something lingering from whatever firmware was sent over on it? I was playing with some can units but it’s working now. If it happens again I won’t reflash. I’ll keeep it and figure out why if it sticks but so far have several airspeed units working.

ryanjAA, 2018-05-26:
So I guess to close this then it’s the argc checks. Maybe we should get that in the docs or add them to all commands.

DanielePettenuzzo, 2018-05-27:
@ryanjAA for the px4 firmware are you using latest master?

ryanjAA, 2018-05-27:
@DanielePettenuzzo Yes. Latest master.

LorenzMeier, 2018-05-27:
@bkueng Sounds like argc handling needs an audit.

dagar, 2018-05-27:
> @bkueng Sounds like argc handling needs an audit.

The driver boilerplate is all copypasta and would greatly benefit from ModuleBase or similar.

ryanjAA, 2018-05-30:
@bkueng I am out of the office for at least another week (no Pix 4 with me either) but so far the airspeed tests with no arguments were doing it. The ver command did not but it probably just needs someone to go through each one with out an argument to see what happens if no other way to check. I’m happy to do so but it won’t be possible for a week or so.

bkueng, 2018-06-04:
@ryanjAA I went through all commands and fixed the ones affected in #9589.

closed 2018-06-04 by LorenzMeier

garfieldG, 2018-06-11:
Hi,

seems like there are some untouched places where the issue isn't handled, such as https://github.com/PX4/Firmware/blob/27e75920bd560b7d8de295db45aba8cf036cd98d/src/drivers/imu/l3gd20/l3gd20.cpp#L1358

bkueng, 2018-06-12:
@garfieldG right, this driver is not enabled on Pixhawk 4. Fixed in #9647