PX4/PX4-Autopilot #9533: General support for rover research platform

LorenzMeier, 2018-05-27:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](slack.px4.io).

**Is your feature request related to a problem? Please describe.**
There is initial support for Rover in PX4 but it needs further polishing and integration. It would be great to have a polished, well-documented setup that suits basic research purposes.

**Describe the solution you'd like**
A complete airframe config for rovers, including a full CI integration tests.

**Additional context**
PX4 can create great value for this application and we need to make it more accessible.

zlite, 2019-01-10:
Sign me up for this one, too

dagar, 2019-08-02:
There's been good progress in this area.

closed 2019-08-02 by dagar