PX4/PX4-Autopilot #9540: FW Mission Start - Wheel increases to max deflection (max pwm value)

ryanjAA, 2018-05-27:
When starting a mission the nose wheel servo instantly increases to ~2000 pwm and there is no time to recover (turns much too much) - need to actually decrease max angles at which the servo can turn but this is a new issue. Here are three logs with this occurring several times during each log. I was able to fly the missions due to some manual offsetting of the wheel but if left centered it would not work every time.

Using an fmu_v2 during these tests. Logging is on from boot so there is a lot of nothing prior.

https://review.px4.io/plot_app?log=02711563-1bd4-40ee-8107-16fa14017b9e

https://review.px4.io/plot_app?log=0285460d-d71f-422b-9d55-b42bccc66cbb

https://review.px4.io/plot_app?log=732a5f97-563d-4141-ae9b-c4c8691277af

ryanjAA, 2018-05-27:
@dagar

ryanjAA, 2018-05-29:
I’d be interested to know if anyone else has experienced anything similar.

LorenzMeier, 2018-05-29:
@ryanjAA We will resolve this within the week as part of the v1.8 sprint.

ryanjAA, 2018-06-03:
@LorenzMeier great! looking forward to testing when resolved.

LorenzMeier, 2018-06-07:
@dagar Could you look into this?

dagar, 2018-06-09:
https://review.px4.io/plot_app?log=0285460d-d71f-422b-9d55-b42bccc66cbb
[image]

@ryanjAA can you try setting FW_W_RMAX to something reasonable (maybe 30 degrees/s)?

ryanjAA, 2018-06-09:
@dagar The first log (https://review.px4.io/plot_app?log=02711563-1bd4-40ee-8107-16fa14017b9e) did have FW_W_RMAX set to 30 (see [and](https://review.px4.io/download?log=02711563-1bd4-40ee-8107-16fa14017b9e&type=1) and was changed at minute 35 to 1 and then 0 to see the result (see attached photo) so the wheel issue was occurring when set like that (at 30). Only changing it to something unlimited (0) was enabling the post manual offset of the servo linkage to then correct for takeoff but this was very much happening at the 30 deg/s. The following logs are flights after the this initial one and show that even when setting to some other number, it is still present.

[image]

ryanjAA, 2018-06-19:
Flew master today and all was fine. 

https://review.px4.io/plot_app?log=e6d82fe3-9b3b-4f4b-b8c6-5af1e930e1a9

closed 2018-06-19 by ryanjAA