PX4/PX4-Autopilot #9549: Tighten GPS fix check

LorenzMeier, 2018-05-28:
**Describe the bug**
The u-blox M8N module over-estimates its accuracy in a multipath or indoor setups.

**To Reproduce**
Run a u-blox M8 inside a house and observe the reported accuracy. 

**Expected behavior**
We need a way to distinguish good reception on a plain field from bad reception inside a house.

**Log Files and Screenshots**

Log file is at the home of the user, so not suitable to share publicly.

[image]

[image]

priseborough, 2018-05-28:
We have a range of EKF internal checks that that can be tightened or enabled. See EKF2_GPS_CHECK parameter which is a bitmask that enables the various checks and the various EKF2_REQ_* parameters that control the check thresholds. 

Some of those checks such as EKF2_REQ_HDRIFT, EKF2_REQ_VDRIFT are not enabled by default because they require the vehicle to be static, but can detect a bad GPS solution independent of the accuracy reported by the receiver.

There are also a number of GPs quality metrics which are reported by the UBlox receiver that we don't use.

thomasgubler, 2018-06-19:
TODO: produce log to be able to tune this. E.g. sport center/gymnasium,warehouse

RomanBapst, 2018-07-03:
I've seen similar behavior when hovering under a tree with both gps and optical flow enabled. In that case the drone drifted around quite a bit and I had to interact. With just optical flow enabled it was rock solid.

RomanBapst, 2018-07-03:
@priseborough @thomasgubler I will produce a replay log hovering under that tree. It has a thick, green canopy.

RomanBapst, 2018-07-10:
@priseborough Here is a replay log of me flying under a tree. The vehicle was actually armed under the tree directly. As you can see on the map from flight review the gps initialised a fair distance away from the tree and then converged to the correct position. When I took off in position control mode the drone started drifting towards the stem immediately and I had to correct via the RC. After a while it stabilized a bit but the horizontal drift was still much bigger than with only optical flow enabled.
https://logs.px4.io/plot_app?log=61bd525d-8527-452a-856c-a8d1ace144bf

priseborough, 2018-07-12:
I've started experimenting with the exisiting preflight GPS checks supported by ekf2 and the filtered position drift levels checked by setting EKF2_GPS_CHECK to 53 were fluctuating between 0.15 and 0.3 m/s. The threshold value controlled by EKF2_REQ_HDRIFT is currently defaulted to 0.3 m/s, but could be reduced to 0.1 m/s. Other parameters that could be changed include:

EKF2_REQ_EPH 5.0 -> 3.0
EKF2_REQ_EPV 8.0 -> 5.0
EKF2_REQ_SACC 1.0 -> 0.5

To prevent the EKF2_REQ_HDRIFT check frustrating users, it will need to be modified to temporarily suspend calculation of the  position derivative when the vehicle is picked up and carried.

thomasgubler, 2018-07-31:
Depending on merge of: https://github.com/PX4/Firmware/pull/9999

closed 2018-08-17 by priseborough