PX4/PX4-Autopilot #9555: Forcing upload of px4fmu-v3_default

pietropier, 2018-05-28:
Hi all, 

by running ver all on mavproxy I verified my Pixhawk doesn't have the 1Mb memory issue. However, when trying to upload the built I get this error:

```
ERROR: 
The Board uses bootloader revision 4 and can therefore not determine
if flashing more than 1 MB (px4fmu-v3_default) is safe, chances are
high that it is not safe! If unsure, use px4fmu-v2_default.

If you know you that the board does not have the silicon errata, use
this script with --force, or update the bootloader.
```

How/where am I supposed to specify this "--force" option?

Thanks in advance

hamishwillee, 2018-05-29:
Good question. This is not documented, but I would *guess* perhaps you can specify it as an option for [uploading](https://dev.px4.io/en/setup/building_px4.html#uploading-firmware-flashing-the-board) when building. 
```
make px4fmu-v2_default upload --force
```
@bkueng Can you confirm or suggest how this might be done? Is there any other way - say through QGC - to force upload of a custom Firmware?

pietropier, 2018-05-29:
I don't know anything about cmake, I tried to put --force in different positions, including the one you suggested but it didn't work. I guess there is some configuration file somewhere, but like I said, I don't know anything about it.

hamishwillee, 2018-05-29:
@pietropier Lets see what @bkueng says. 

Note however this is not the best place for [support requests](https://docs.px4.io/en/#support). I'm only continuing the discussion here because the docs are clearly broken in this respect. You might want to ask on the general channel on slack.

pietropier, 2018-05-29:
I was able to upload the built using the custom firmware upload function in QGC. However, if there is a more direct solution to force the upload through make I'm still interested in it. 

@hamishwillee I will post this on Slack as well, thanks.

WrongChao, 2018-05-29:
1.Open Firmware/platforms/nuttx/CMakeLists.txt.
2.Change "${PX4_SOURCE_DIR}/Tools/px_uploader.py --port ${serial_ports} ${fw_file}" (line:178) to "${PX4_SOURCE_DIR}/Tools/px_uploader.py --force --port ${serial_ports} ${fw_file}".
3.Now retry to make and upload.

Or you can change the file Firmware/Tools/px_uploader.py too:
1.Find the function "def upload(self, fw, force=False, boot_delay=None)"
2.Change the default value of the param "force" to "True".

bkueng, 2018-05-29:
> Can you confirm or suggest where --force is used?

After building the firmware you do:
`./Tools/px_uploader.py build/px4fmu-v3_default/px4fmu-v3_default.px4 --port /dev/ttyACM0 --force`
You may need to adjust the port.

> Are there instructions anywhere for updating the bootloader?

Not yet. The goal is to implement a user-friendly method in QGC (https://github.com/mavlink/qgroundcontrol/issues/6488).

hamishwillee, 2018-11-21:
You can now update the bootloader - see https://docs.px4.io/en/advanced_config/bootloader_update.html

This puts a new bootloader on. The new bootloader is able to tell if the vehicle can be updated with FMUv3. It should be safe to run even if the board does have the silicon errata.

@bkueng This can be closed now right?

bkueng, 2018-11-21:
I'd say so.

closed 2018-11-21 by bkueng