PX4/PX4-Autopilot #9557: A java problem with make posix jmavsim in Ubuntu

Hs293Go, 2018-05-28:
Hello everybody, I hit a snag with jmavsim SITL simulation on ubuntu but I tried to strictly follow instructions. I run Ubuntu 18.04 (Bionic), gazebo 9 and ROS Melodic. Firmware and submodules are up to date.

My problem is that `make posix jmavsim` results in an error as shown below. 

> BUILD SUCCESSFUL
> Total time: 0 seconds
> Exception in thread "main" java.lang.NoClassDefFoundError: javax/vecmath/Tuple3d
> 	at java.base/java.lang.Class.forName0(Native Method)
> 	at java.base/java.lang.Class.forName(Class.java:374)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)
> Caused by: java.lang.ClassNotFoundException: javax.vecmath.Tuple3d
> 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:466)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:566)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
> 	... 3 more
> Exception in thread "main" java.lang.NoClassDefFoundError: javax/vecmath/Tuple3d
> 	at java.base/java.lang.Class.forName0(Native Method)
> 	at java.base/java.lang.Class.forName(Class.java:374)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)
> Caused by: java.lang.ClassNotFoundException: javax.vecmath.Tuple3d
> 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:466)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:566)
> 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:499)
> 	... 3 more

I believe this is not a new issue, as it appeared to be almost identical to [#9311](https://github.com/PX4/Firmware/issues/9311), with the exception that I use ubuntu. My Java version is 

> openjdk version "10.0.1" 2018-04-17
> OpenJDK Runtime Environment (build 10.0.1+10-Ubuntu-3ubuntu1)
> OpenJDK 64-Bit Server VM (build 10.0.1+10-Ubuntu-3ubuntu1, mixed mode)

Per @julianoes 's suggestion, switching to Java 8 (either java-8-openjdk-amd64 or java-8-oracle) and redoing the build resulted in the following:

> BUILD SUCCESSFUL
> Total time: 1 second
> Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0
> 	at java.lang.ClassLoader.defineClass1(Native Method)
> 	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
> 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
> 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
> 	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
> 	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
> 	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
> 	at java.lang.Class.forName0(Native Method)
> 	at java.lang.Class.forName(Class.java:348)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)
> Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0
> 	at java.lang.ClassLoader.defineClass1(Native Method)
> 	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
> 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
> 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
> 	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
> 	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
> 	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
> 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
> 	at java.lang.Class.forName0(Native Method)
> 	at java.lang.Class.forName(Class.java:348)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)

Strangely enough, gazebo setup is largely successful and I am able to simulate in both SITL and HITL. Nevertheless I could really use jmavsim as a lighter, faster simulator since I am only concerned with multirotors. Thank you for looking into this issue!

hamishwillee, 2018-05-29:
1. What version of ubuntu are you using?
2. Which script did you use to set things up? The one here? https://dev.px4.io/en/setup/dev_env_linux.html#gazebo-with-ros
3. Are you on a real computer or on a VM?

I work primarily on a Ubuntu VM in Windows and have historically been using jMAVSim for quick tests. Unfortunately this no longer works in the VM, so I feel your pain. That said, I have found Gazebo running headless to be similarly performant to jMAVSim (slower to build first time, but about the same when running). To do this you basically prefix your make command with `HEADLESS=1 make ...`

Hs293Go, 2018-05-29:
I am using a real computer, and ROS is less of a problem: Since I can appreciate the delicate link between ros and ubuntu distros, I rely on another computer running ubuntu 16.04 and Kinetic for ros.

The trouble with jmavsim is that I can't isolate the issue. Is it a problem with java? Incompatible ubuntu distro? I'd greatly appreciate some advice to point me in the direction of a diagnosis.

hamishwillee, 2018-05-29:
The second error looks like you have a mismatched JDK and JRE - ie you're running jMAVSIM compiled using a JDK later than 8. Did you do a clean build before trying?

Other than that I don't know. Julian's suggestion of Java 8 is what we use in our docker environments, and I think is the likely pain point. 

I'd love this fixed but jMAVSim problems aren't IMO treated as high priority as say Gazebo ones. So if you can work with Gazebo headless that would be smart.

Hs293Go, 2018-05-29:
I proceeded with deleting all of `out/` and rebuilded. 'java --version' shows the following

> openjdk version "1.8.0_171"
> OpenJDK Runtime Environment (build 1.8.0_171-8u171-b11-0ubuntu0.18.04.1-b11)
> OpenJDK 64-Bit Server VM (build 25.171-b11, mixed mode)

Build is not successful and shows the following

> Exception in thread "main" java.lang.reflect.InvocationTargetException
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
> Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
> 	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
> 	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
> 	at java.awt.Window.getToolkit(Window.java:1358)
> 	at java.awt.Window.init(Window.java:506)
> 	at java.awt.Window.<init>(Window.java:537)
> 	at java.awt.Frame.<init>(Frame.java:420)
> 	at java.awt.Frame.<init>(Frame.java:385)
> 	at javax.swing.JFrame.<init>(JFrame.java:189)
> 	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:104)
> 	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
> 	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
> 	... 5 more
> Options parsed, starting Sim.
> Starting GUI...
> Exception in thread "main" java.lang.reflect.InvocationTargetException
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
> Caused by: java.awt.AWTError: Assistive Technology not found: org.GNOME.Accessibility.AtkWrapper
> 	at java.awt.Toolkit.loadAssistiveTechnologies(Toolkit.java:807)
> 	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:886)
> 	at java.awt.Window.getToolkit(Window.java:1358)
> 	at java.awt.Window.init(Window.java:506)
> 	at java.awt.Window.<init>(Window.java:537)
> 	at java.awt.Frame.<init>(Frame.java:420)
> 	at java.awt.Frame.<init>(Frame.java:385)
> 	at javax.swing.JFrame.<init>(JFrame.java:189)
> 	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:104)
> 	at me.drton.jmavsim.Simulator.<init>(Simulator.java:157)
> 	at me.drton.jmavsim.Simulator.main(Simulator.java:678)
> 	... 5 more

Is using **openjdk** instead of **oracle java** likely to cause this issue? I might not be able to rebuild with oracle-8-java just yet because of some minor problems.

hamishwillee, 2018-05-29:
> Is using openjdk instead of oracle java likely to cause this issue? I might not be able to rebuild with oracle-8-java just yet because of some minor problems.

Now, we use open jdk in our docker setup.

TSC21, 2018-05-29:
> @TSC21 Has work been done to validate/test PX4 and MAVRos with ROS Melodic?

MAVROS is already available for ROS Melodic.

hamishwillee, 2018-05-30:
@TSC21 Thanks! Is it time to update our docs/install scripts etc to target ROS Melodic (instead of Kinetic) - or is there a need for both/it is too early?

Hs293Go, 2018-05-30:
Okay, the problem is solved by using the following version of java. 
> java version "1.8.0_171"
> Java(TM) SE Runtime Environment (build 1.8.0_171-b11)
> Java HotSpot(TM) 64-Bit Server VM (build 25.171-b11, mixed mode)

Build succeeded and jmavsim is up and running. One question remaining: Why does Oracle java succeed while Openjdk fail?

Paschy11, 2018-06-21:
Hi i got the same error and i have these versions of java
openjdk version "1.8.0_171"
OpenJDK Runtime Environment (build 1.8.0_171-8u171-b11-0ubuntu0.17.10.1-b11)
OpenJDK 64-Bit Server VM (build 25.171-b11, mixed mode)
....
How did you solve that?

julianoes, 2018-06-25:
@Paschy11 it looks like you need to use Oracle Java and not OpenJDK.

closed 2018-06-25 by julianoes

Paschy11, 2018-06-26:
I'm using Oracle java right now and i have these:
java version "1.8.0_171"
Java(TM) SE Runtime Environment (build 1.8.0_171-b11)
Java HotSpot(TM) 64-Bit Server VM (build 25.171-b11, mixed mode)
exactly the same ones Hs293Go succeeded with. Even if i'm using this version of Java Runtime i'm having this error when i try to launch "make posix jmavsim" after cancelling everything in the out folder:
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)
Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:56)
Any Ideas?

Hs293Go, 2018-06-26:
> Exception in thread "main" java.lang.UnsupportedClassVersionError: me/drton/jmavsim/Simulator has been compiled by a more recent version of the Java Runtime (class file version 54.0), this version of the Java Runtime only recognizes class file versions up to 52.0

I ran into this too. In my case I didn't do a clean build. Deleting all of `out/` solved the problem for me.

On a side note, I've been using ROS Melodic with PX4/MAVROS with no problem. Let me know if there's anything I can do to help with the migration to Melodic

julianoes, 2019-07-17:
I just ran into this again. Oracle Java is not needed, OpenJDK 8 is fine.
However, this workaround is needed:

```
sudo gedit /etc/java-8-openjdk/accessibility.properties
```

and comment out the following line like this:
```
#assistive_technologies=org.GNOME.Acessibility.AtkWrapper
```

Fix found here: https://askubuntu.com/questions/695560

hamishwillee, 2019-07-18:
@julianoes Does this affect dev guide instructions/scripts?
- http://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html#jmavsim
- https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_common_deps.sh
- https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh

julianoes, 2019-07-18:
@hamishwillee we could add a note in a troubleshooting section

julianoes, 2019-07-30:
> @hamishwillee we could add a note in a troubleshooting section

Done: https://github.com/PX4/Devguide/pull/846

hamishwillee, 2019-07-30:
Thanks @julianoes . Merged. I was aware of that, but it was still sitting on my todo list.  Much appreciated.

Bikappa, 2019-09-17:
@julianoes thanks, it worked for me

rk800506, 2020-02-21:
> I just ran into this again. Oracle Java is not needed, OpenJDK 8 is fine.
> However, this workaround is needed:
>
> ```
> sudo gedit /etc/java-8-openjdk/accessibility.properties
> ```
>
> and comment out the [...]

The line is already commented out, but the error is still there

julianoes, 2020-02-21:
https://dev.px4.io/master/en/simulation/jmavsim.html#troubleshooting

Do this before trying again:
```
rm -rf Tools/jMAVSim/out
```