PX4/PX4-Autopilot #9561: MAV_CMD_SET_MESSAGE_INTERVAL.param2 = 0 stops stream instead of reset to default

DonLakeFlyer, 2018-05-30:
I'm testing using SITL and firmware master. For the new QGC PID tuning support I speed up the rates for MAVLINK_MSG_ID_ATTITUDE and MAVLINK_MSG_ID_ATTITUDE_TARGET using the MAV_CMD_SET_MESSAGE_INTERVAL command. When I up the rates it works. But when I want to reset the rates back to default they instead fall to a rate of 0.

DonLakeFlyer, 2018-05-30:
From MavlinkReceiver::set_message_interval(int msgId, float interval, int data_rate)
```
	} else {
		// note: mavlink spec says rate == 0 is requesting a default rate but our streams
		// don't publish a default rate so for now let's pick a default rate of zero.
	}
```

Uh, not super useful. Also in looking at the code it is passing something in from param3 which is not in the mavlink spec.

LorenzMeier, 2018-05-30:
@DonLakeFlyer Please use ATTITUDE_QUATERNION instead because ATTITUDE suffers from gimbal lock and will be deprecated. In terms of other messages: LOCAL_POSITION and the associated _TARGET message will likely be helpful to tune the position controller.

DonLakeFlyer, 2018-05-30:
> Please use ATTITUDE_QUATERNION instead because ATTITUDE suffers from gimbal lock and will be deprecated.

Got it. QGC code needs a major cleanup and how it uses the various messages which can have attiude information in it: ATTITUDE, ATTITUDE_QUATERNION, VFR_HUD. It's a bit crazy not due to likely back compat for realy really old firmwares.

LorenzMeier, 2018-06-15:
@bkueng Do you see a quick fix for this?

bkueng, 2018-06-15:
QGC could use `MAV_CMD_GET_MESSAGE_INTERVAL` beforehand?

Otherwise we could do some refactoring and call https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2005:L2033 again, but only applied to the requested stream. This would avoid any memory overhead.

DonLakeFlyer, 2018-06-15:
Would be nice if it worked as spec'ed.

PR #9759 (2018-06-25): Mavlink stream updates
- improve mavlink status output (add datarate, UDP remote port and partner IP)
- only enable HIL_ACTUATOR_CONTROLS in hil if link has enough bandwidth
- allow resetting mavlink streams to default rate via MAV_CMD_SET_MESSAGE_INTERVAL (@DonLakeFlyer )
- add a `mavlink status streams` command to display all configured streams. Example output:
```
instance #1:
        Name                Rate Config (current) [Hz] Message Size (if active) [B]
        HEARTBEAT                       1.00 (1.000)    21
        STATUSTEXT                     20.00 (6.320)
        COMMAND_LONG  [... cut]
closed 2018-06-25 by LorenzMeier