PX4/PX4-Autopilot #9566: Feature request: new COM_ARM_ASPD parameter

Antiheavy, 2018-05-30:
This request is to add a new parameter to prevent arming if airspeed is reading too high (or too low).  Different vehicles and use cases require different thresholds so it should be a parameter.

If the PX4 diff pressure sensor module is changed allowed to allow sending negative airspeed to the rest of the system, then this threshold should be a +/- value, meaning if too large of a negative value is being read, then also prevent arming (because there may be a blockage or calibration problem).

Set to 0 to disable.

Send appropriate warning/error messages if arming is prevented due to airspeed reading.

Here is a flight log where a fixed wing vehicle was allowed to arm and takeoff with a pinched diff pressure hose, ultimately resulting in a crash. A check like this at arming would have caught this issue.
https://review.px4.io/plot_app?log=9677a79d-1722-4f95-9649-deb3efc84786

Antiheavy, 2018-06-06:
If hard coded what would be a value that would work for all/most use cases?  I’m not sure I know.   I guess we could also ask ourselves if just forcing the standard diff pressure preflight check at time of Arming would be good enough. Again, not sure. Might need to play around and see.

dagar, 2018-06-08:
I'm not sure, right now we have a hard coded 15 Pascal differential pressure sensor check.

Antiheavy, 2018-06-10:
@Kjkinney let's ground test this soon.

Antiheavy, 2018-06-14:
@dagar 
after our discussion, it makes sense to combine this with the existing pre-flight check for Diff-Pressure and also run this check at ARM.   Change the check to look at airspeed instead of diff pressure. This requires allowing negative airspeed to pass into the system as well.

Antiheavy, 2018-06-25:
testing here.  https://github.com/SenteraLLC/PX4-Firmware/issues/116

Antiheavy, 2018-07-09:
related to this: https://github.com/PX4/Firmware/pull/9636

Antiheavy, 2018-08-30:
one negative consequence of this with the current architecture is that you will not be able to disarm and then re-arm in flight.  There are other problems assocaited with disarming and re-arming in flight anyway, such as resetting the field attitude which I think screws up auto landing.  Maybe that is a separate issue.

Antiheavy, 2019-01-22:
@dagar we've been using a hard coded solution to this for a while now and it works fine.  We should discuss if/how it could be adapted for upstream.

closed 2019-07-08 by stale[bot]