PX4/PX4-Autopilot #9567: allow system to show negative airspeed

Antiheavy, 2018-05-30:
Currently PX4 does not pass though negative airspeed (or diff pressure?) measurements from the diff pressure sensor module.  This hides potential problems from the users such as:  bad calibration or blocked/pinched or reversed pitot-static lines.  Negative values should be allowed to be sent to the system and ultimately displayed to the user.  These negative values can also be used to detect problems or prevent arming in other parts of the system.

partially related issue: https://github.com/PX4/Firmware/issues/9566

Antiheavy, 2018-06-10:
@Kjkinney let's ground test this soon.

dagar, 2018-06-11:
PR https://github.com/PX4/Firmware/pull/9636

Antiheavy, 2018-06-25:
I think this is working fine in the recent Sentera builds.  I'll close it and we will continue to roll it in as a separate add-on to the sentera builds until it makes it into upstream and we sync.

closed 2018-06-25 by Antiheavy