PX4/PX4-Autopilot #9568: not able to launch posix_sitl.launch

zubair123khan, 2018-05-30:
Hi 

I have followed the instruction from here https://dev.px4.io/en/simulation/ros_interface.html and after successful build i did roslaunch px4 posix_sitl.launch  and its showing me : 
ResourceNotFound: gazebo_ros
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/zubair/px4_ws/src/mavlink
ROS path [2]=/home/zubair/px4_ws/src/mavros/libmavconn
ROS path [3]=/home/zubair/px4_ws/src/mavros/mavros_msgs
ROS path [4]=/home/zubair/px4_ws/src/mavros/mavros
ROS path [5]=/home/zubair/px4_ws/src/mavros/mavros_extras
ROS path [6]=/home/zubair/px4_ws/src/mavros/test_mavros
ROS path [7]=/opt/ros/kinetic/share
ROS path [8]=/home/zubair/src/Firmware
ROS path [9]=/home/zubair/src/Firmware/Tools/sitl_gazebo
ROS path [10]=/home/zubair/Firmware
ROS path [11]=/home/zubair/Firmware/Tools/sitl_gazebo

I am using ROS Kinetic by the way
can anyone help me with this ??

hamishwillee, 2018-05-30:
What Ubuntu platform are you running on and what script did you use to set up the environment?

zubair123khan, 2018-05-31:
Thanks, I just solved the problem by using "aptitude" and by installing ros
packages again using aptitude as I had broken packages..

To people facing the same problem.

lamping7, 2018-05-31:
Yes. gazebo_ros is the package that allows Gabezo to interface with ROS. Possibly you didn't install ROS with `desktop-full`, which contains this package.

closed 2018-05-31 by lamping7