PX4/PX4-Autopilot #9569: Simulator gives MANUAL CONTROL LOST

hamishwillee, 2018-05-31:
When running simulator there are lots of warnings “Manual Control Lost”  and shortly afterwards “Manual Control Regained”. We shouldn't be getting these warnings because we don't have manual control (Simulator) and COM_RC_IN_MODE is set by default such that RC fails should be ignored.

This can be replicated Gazebo Standard VTOL (Macbook pro) and on jmavsim or Gazebo multicopter on Linux computer running Ubuntu 16.04.  

I can reproduce on Ubuntu on any build in the last couple of weeks. Simply start the simulator and then putting my computer into lock mode. After COM_RC_LOSS_T  the manual control lost message is output in QGC. Undoing lock regains control. I THINK this happens at other times, but I cannot reporduce that. 

FYI @RomanBapst. Reported by Ivo Dresher on slack here: https://px4.slack.com/archives/C0V533X4N/p1527595988000400

[image]

hamishwillee, 2018-05-31:
A temporary workaround is to set COM_RC_LOSS_T timeout to maximum.

ivodre, 2018-05-31:
Thanks Hamish for creating the defect report. For me setting NAV_RCL_ACT to "RC Auto Recovery" helped. With that setting when MANUAL CONTROL LOST happened this had no influence on the simulation. I had COM_RC_LOSS_T at 0.5s all the time.

closed 2019-07-09 by stale[bot]

julianoes, 2019-07-09:
I believe this is fixed, please re-open if it's still an issue.

echoGee, 2019-07-12:
@julianoes : On which version is this fixed ? Is this fixed without setting COM_RC_LOSS_T timeout to maximum? 
I can still reproduce this issue

julianoes, 2019-07-15:
@echoGee
> Simply start the simulator and then putting my computer into lock mode.

What do you mean lock mode?

echoGee, 2019-07-15:
I am not sure what is Iock mode, assumed it meant hold mode. But I can reproduce the issue in stabilized or hold mode.

julianoes, 2019-07-18:
@echoGee 

I'm confused. I don't see that and I use SITL/HITL all the time. Is it SITL or HITL?

@hamishwillee what do you mean with?

> What do you mean lock mode?

hamishwillee, 2019-07-18:
@julianoes 
By lock mode, I mean Windows screen lock mode. This can be invoked by doing **Ctrl + Alt + Delete** and then selecting Lock. Or it happens after a computer has been left alone for a timeout. 

I haven't seen this issue for a while, but it may be because I disabled automatic locking of my computer when I am absent.

julianoes, 2019-07-18:
Aha, I'd say that's expected because the OS basically stops and the time slips. With lockstep it should not happen anymore though.

hamishwillee, 2019-07-19:
IN that case, from my perspective this is closed.

echoGee, 2019-07-22:
I can reproduce this on Ubuntu 16.04 with docker. Is it due to too much processor hogging ?

julianoes, 2019-07-22:
@echoGee what PX4 version?

echoGee, 2019-07-28:
Version is 1.8.2

julianoes, 2019-07-29:
Right, in that case it's probably because the simulation runs too slowly on your computer. I can only recommend to use 1.9 where we have lockstep for exactly this problem.