PX4/PX4-Autopilot #9572: Pixhawk mini. PREFLIGHT FAIL: NO MAG SENSOR 0

Texnovik, 2018-05-31:
Hi

I have:
PixhavkMini Flight Controller
QGroundCOntrol v3.3.2
In the settings of the Airframe Setup choose DJI Flame Wheel F330 (tried other options)

I do the installation of the firmware of the instruction. Firmware version 1.7.3
At the calibration stage of the sensors I get an error PREFLIGHT FAIL: NO MAG SENSOR 0

Repeat the installation several times. Tried to reset all parameters. Nothing helps.

What could be the problem ? I spent several days on her decision ... Nothing works.

bkueng, 2018-05-31:
Hi

Can you provide the output of the following commands:
- `sensors status`
- `ls /dev`
- `listener sensor_mag`

mhkabir, 2018-05-31:
@Texnovik Do you have a GPS-Compass unit connected? Pixhawk Mini does not have a usable internal compass.

Texnovik, 2018-05-31:
@mhkabir,  I know that Pixmini does not have a internal compass. I use the standard compass 3DR from the kit. Strange... but from the firmware of ArduPilot it works ...

Texnovik, 2018-05-31:
@bkueng, I know little about the console in MAC OS... command result:

MacMini:~ Olegov$ sensors status
-bash: sensors: command not found

MacMini:~ Olegov$ ls /dev
HAX				osxfuse59			ptyu5				ttyr1
afsc_type5			osxfuse6			ptyu6				ttyr2
auditpipe			osxfuse60			ptyu7				ttyr3
auditsessions			osxfuse61			ptyu8				ttyr4
autofs				osxfuse62			ptyu9				ttyr5
autofs_control			osxfuse63			ptyua				ttyr6
autofs_homedirmounter		osxfuse7			ptyub				ttyr7
autofs_notrigger		osxfuse8			ptyuc				ttyr8
autofs_nowait			osxfuse9			ptyud				ttyr9
bpf0				pf				ptyue				ttyra
bpf1				pfm				ptyuf				ttyrb
bpf2				pmCPU				ptyv0				ttyrc
bpf3				profile				ptyv1				ttyrd
console				ptmx				ptyv2				ttyre
cu.Bluetooth-Incoming-Port	ptyp0				ptyv3				ttyrf
cu.usbmodem1			ptyp1				ptyv4				ttys0
disk0				ptyp2				ptyv5				ttys000
disk0s1				ptyp3				ptyv6				ttys1
disk0s2				ptyp4				ptyv7				ttys2
disk1				ptyp5				ptyv8				ttys3
disk1s1				ptyp6				ptyv9				ttys4
disk1s2				ptyp7				ptyva				ttys5
disk1s3				ptyp8				ptyvb				ttys6
disk2				ptyp9				ptyvc				ttys7
disk2s1				ptypa				ptyvd				ttys8
disk2s2				ptypb				ptyve				ttys9
disk2s3				ptypc				ptyvf				ttysa
disk2s4				ptypd				ptyw0				ttysb
disk3				ptype				ptyw1				ttysc
disk3s1				ptypf				ptyw2				ttysd
dtrace				ptyq0				ptyw3				ttyse
dtracehelper			ptyq1				ptyw4				ttysf
fbt				ptyq2				ptyw5				ttyt0
fd				ptyq3				ptyw6				ttyt1
fsevents			ptyq4				ptyw7				ttyt2
io8log				ptyq5				ptyw8				ttyt3
io8logmt			ptyq6				ptyw9				ttyt4
io8logtemp			ptyq7				ptywa				ttyt5
klog				ptyq8				ptywb				ttyt6
lockstat			ptyq9				ptywc				ttyt7
machtrace			ptyqa				ptywd				ttyt8
null				ptyqb				ptywe				ttyt9
oslog				ptyqc				ptywf				ttyta
oslog_stream			ptyqd				random				ttytb
osxfuse0			ptyqe				rdisk0				ttytc
osxfuse1			ptyqf				rdisk0s1			ttytd
osxfuse10			ptyr0				rdisk0s2			ttyte
osxfuse11			ptyr1				rdisk1				ttytf
osxfuse12			ptyr2				rdisk1s1			ttyu0
osxfuse13			ptyr3				rdisk1s2			ttyu1
osxfuse14			ptyr4				rdisk1s3			ttyu2
osxfuse15			ptyr5				rdisk2				ttyu3
osxfuse16			ptyr6				rdisk2s1			ttyu4
osxfuse17			ptyr7				rdisk2s2			ttyu5
osxfuse18			ptyr8				rdisk2s3			ttyu6
osxfuse19			ptyr9				rdisk2s4			ttyu7
osxfuse2			ptyra				rdisk3				ttyu8
osxfuse20			ptyrb				rdisk3s1			ttyu9
osxfuse21			ptyrc				sdt				ttyua
osxfuse22			ptyrd				stderr				ttyub
osxfuse23			ptyre				stdin				ttyuc
osxfuse24			ptyrf				stdout				ttyud
osxfuse25			ptys0				systrace			ttyue
osxfuse26			ptys1				tty				ttyuf
osxfuse27			ptys2				tty.Bluetooth-Incoming-Port	ttyv0
osxfuse28			ptys3				tty.usbmodem1			ttyv1
osxfuse29			ptys4				ttyp0				ttyv2
osxfuse3			ptys5				ttyp1				ttyv3
osxfuse30			ptys6				ttyp2				ttyv4
osxfuse31			ptys7				ttyp3				ttyv5
osxfuse32			ptys8				ttyp4				ttyv6
osxfuse33			ptys9				ttyp5				ttyv7
osxfuse34			ptysa				ttyp6				ttyv8
osxfuse35			ptysb				ttyp7				ttyv9
osxfuse36			ptysc				ttyp8				ttyva
osxfuse37			ptysd				ttyp9				ttyvb
osxfuse38			ptyse				ttypa				ttyvc
osxfuse39			ptysf				ttypb				ttyvd
osxfuse4			ptyt0				ttypc				ttyve
osxfuse40			ptyt1				ttypd				ttyvf
osxfuse41			ptyt2				ttype				ttyw0
osxfuse42			ptyt3				ttypf				ttyw1
osxfuse43			ptyt4				ttyq0				ttyw2
osxfuse44			ptyt5				ttyq1				ttyw3
osxfuse45			ptyt6				ttyq2				ttyw4
osxfuse46			ptyt7				ttyq3				ttyw5
osxfuse47			ptyt8				ttyq4				ttyw6
osxfuse48			ptyt9				ttyq5				ttyw7
osxfuse49			ptyta				ttyq6				ttyw8
osxfuse5			ptytb				ttyq7				ttyw9
osxfuse50			ptytc				ttyq8				ttywa
osxfuse51			ptytd				ttyq9				ttywb
osxfuse52			ptyte				ttyqa				ttywc
osxfuse53			ptytf				ttyqb				ttywd
osxfuse54			ptyu0				ttyqc				ttywe
osxfuse55			ptyu1				ttyqd				ttywf
osxfuse56			ptyu2				ttyqe				urandom
osxfuse57			ptyu3				ttyqf				zero
osxfuse58			ptyu4				ttyr0

MacMini:~ Olegov$ listener sensor_mag
-bash: listener: command not found

... probably I'm doing it wrong :(

bkueng, 2018-05-31:
You should use the console in QGC: Go to Analyze (https://docs.qgroundcontrol.com/en/toolbar/toolbar.html), and then Mavlink Console.

Texnovik, 2018-05-31:
@bkueng 
[image]

Texnovik, 2018-05-31:
@bkueng, I'm sorry ... I'm inattentive ... Do these commands depend on the firmware (PX4 or ArduPilot) ? I now have a ArduPilot.

mhkabir, 2018-05-31:
These commands are for PX4.

Texnovik, 2018-05-31:
[image]

Texnovik, 2018-05-31:
[image]

bkueng, 2018-06-01:
Thanks @Texnovik. There is no compass driver running. Can you find out which compass is installed on your GPS device?

Texnovik, 2018-06-01:
@bkueng, Hello ! Photo of the compass ...
[image]
[image]

bkueng, 2018-06-01:
From the pictures I don't see which compass is on there. Do you have a link from where you bought it?

You can try to start this drivers: `ist8310 -C -R 4 start`, and then check again with `listener sensor_mag` if a compass got detected.

Texnovik, 2018-06-01:
I bought it from Aliexpress. I'm confused by some details ... On the box Pixhawk written holibro.com. On this site there is no Pixhawk, but there is a PixFalcon ... Maybe a PixFalcon is installed inside my Pixhawk? (this is a joke) :) On this site I found a GPS like mine [http://www.holybro.com/product/22](url). Maybe this will help ?..

command result

[image]

Texnovik, 2018-06-03:
Hello ! I found this article http://discuss.px4.io/t/pin2pin-alternative-replacement-to-honeywell-hmc5883l-hmc5983-eol-compass/944. Probably I have an old model of the compass hmc5883. I tried to run the command hmc5883 -C -T -X start. The result is below. Probably this is the problem.

[image]

Texnovik, 2018-06-10:
The problem is solved. The compass was defective. I installed a new compass. Everything is working. Thank you all for your help!

closed 2018-06-10 by Texnovik