PX4/PX4-Autopilot #9573: Quaternion computation doubts in attitude control

xiongmaoer, 2018-05-31:
I'm trying to understand the paper “Nonlinear Quadrocopter Attitude Control (2013)” referred by Pix4 in mc_att_control_main.cpp.I have a basic knowledge of quaterions and rotations,but i'm confused now.

1.in the paper Section 3.2.1 reduced attitude control , we can see formula (45) and (47):
[image]
the qe,red refers to the rotation form current body attitude q to the reduced attitude qcmd,red , so ,why is vector k expressed in world frame I ? We don't rotate the coordinate system by vector k using expressions based on world frame I,actually we rotate the coordinate system just based on the current attitude q.In my point of view, i think k should be rewritten in body frame.
[image]

2.`/* calculate reduced desired attitude neglecting vehicle's yaw to prioritize roll and pitch */
	Vector3f e_z = q.dcm_z();
	Vector3f e_z_d = qd.dcm_z();
	Quatf qd_red(e_z, e_z_d);

	if (abs(qd_red(1)) > (1.f - 1e-5f) || abs(qd_red(2)) > (1.f - 1e-5f)) {
		/* In the infinitesimal corner case where the vehicle and thrust have the completely opposite direction,
		 * full attitude control anyways generates no yaw input and directly takes the combination of
		 * roll and pitch leading to the correct desired yaw. Ignoring this case would still be totally safe and stable. */
		qd_red = qd;

	} else {
		/* transform rotation from current to desired thrust vector into a world frame reduced desired attitude */
		qd_red *= q;
	}`
In last line we can see qd_red=qd_red*q , which can be seen as qd_red =qd_red * q,however in formula (47) the q is right-multiplied by qe,red . Maybe qd_red=q*qd_red is right , for quaterions don't satisfy commutative law.I hope the authors of the paper can see my questions.

MaEtUgR, 2018-06-01:
@xiongmaoer Let me try to help you. The authors of the paper do probably not see your question. I implemented the quaternion controller in PX4 and referenced their publication because it explains the principles of such a controller in a formal way.

1. Still thinking about the details on how to explain, it's not trivial. I'll write when I wrapped up.
2. Yes, both are correct, the difference is the used quaternion definition. As you likely already know people can historically not agree on one quaternion convention. PX4 uses the **Hamilton quaternion convention** which is implemented in the https://github.com/PX4/Matrix library since https://github.com/PX4/Matrix/pull/34.  We agreed on it because it has a known name, is clearly defined and widely used in other libraries like eigen. Sadly the referenced paper uses its own convention which only partly matches. Note that in a specific case the multiplication order does not only depend on the algebraic definition of the quaternion multiplication itself but also on the interpretation of each quaternion. If I remember correctly the algebraic multiplication (or handedness) of the paper is the same as with Hamilton but the interpretation for rotating a vector is world to body which is opposite to Hamilton. If you want to read further on that subject I can strongly recommend chapter 3 (page 33) of this paper: http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf

xiongmaoer, 2018-06-01:
@MaEtUgR Thanks a lot for your help . I really din’t consider from this aspect . Maybe i should learn more about quaternion and i will see again how it's implemented in Matrix library. 
As for my question 1, the k vector still confused me .When we computate a quaternion form world frame e to body frame b ，using the euler angles in the ZYX order：
[image]
[image]
Here i consider the second quaternion qy（Theta），obviously by the definition of quaternion ，it means a rotation around k2(n2) vector with an angle of Theta rad, while the k2 vector is [0 1 0]T,which
is expressed in the k1k2k3 frame in picture (b). So let's go back to the question in formula (45), i can't
understand why the k vector isn't expressed in body frame.

weihli, 2018-07-17:
@xiongmaoer , for question 1, both sides of the equation, left multiply inv(q); 
we can get : q^(-1)*q_(cmd,red)=q_(e,red)

weihli, 2018-08-01:
@xiongmaoer  Your two questions also comused me before I read the papers : "Quaernions and attitude representation" and  “Introduction into quaternions for spacecraft attitude representation”。 In the code Quatf qd_red(e_z, e_z_d); qd_red is rotation from current z axis to reduced z axis, but it is express on world frame as your question 1;  so we can transformate it to current body frame : qd_red_body = inv(q) *  qd_red * q;  qd_red_body is the rotation from body to reduced which expression on the body frame; now  we can get new qd_red which rotate from world to reduced: qd_red = q * qd_red_body = q * inv(q) * qd_red * q = qd_red * q (same as the code qd_red *= q)

xiongmaoer, 2019-01-20:
> @xiongmaoer Your two questions also comused me before I read the papers : "Quaernions and attitude representation" and “Introduction into quaternions for spacecraft attitude representation”。 In the [...]

you are right

MaEtUgR, 2019-01-26:
I hope it is clearer now, sorry I didn't find the words for a better explanation. But I wrote a python script to experiment and visualize the calculations here: https://github.com/Auterion/Flight_Control_Prototyping_Scripts/tree/master/quaternion_attitude_control

Feel free to try it out and ask further if you still feel it's not correct.

closed 2019-01-26 by MaEtUgR

lixiaowei123, 2024-02-21:
> @xiongmaoer Your two questions also comused me before I read the papers : "Quaernions and attitude representation" and “Introduction into quaternions for spacecraft attitude representation”。 In the [...]

qd_red = q * qd_red_body = q * inv(q) * qd_red * q = qd_red * q (same as the code qd_red *= q) Do you have any theoretical basis in this field