PX4/PX4-Autopilot #9575: Error in make posix_sitl_default gazebo

akshaykvnit, 2018-05-31:
I am performing simulation of a quadcopter with Pixhawk in offboard mode. I was following [this link](https://dev.px4.io/en/simulation/gazebo.html).

When I do 
`make posix_sitl_default`

I get the following
```
-- PX4 VERSION: v1.8.0-beta2-16-g6ca0784
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-521-ga80b3ab
-- Configuring done
-- Generating done
-- Build files have been written to: /home/akshay/src/Firmware/build/posix_sitl_default
[562/632] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.83
-- Found Simbody: /usr/include/simbody  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so  
-- Boost version: 1.58.0
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
[... 133 lines cut ...]
[42/72] Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
[43/72] Building CXX object CMakeFiles/rotors_gazebo_imu_plugin.dir/src/gazebo_imu_plugin.cpp.o
[44/72] Building CXX object CMakeFiles/rotors_gazebo_multirotor_base_plugin.dir/src/gazebo_multirotor_base_plugin.cpp.o
[45/72] Building CXX object CMakeFiles/rotors_gazebo_controller_interface.dir/src/gazebo_controller_interface.cpp.o
[46/72] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o
[47/72] Building CXX object CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o
[48/72] Building CXX object CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/geo_mag_declination.cpp.o
[49/72] Building CXX object CMakeFiles/gazebo_irlock_plugin.dir/src/gazebo_irlock_plugin.cpp.o
[50/72] Building CXX object CMakeFiles/rotors_gazebo_gimbal_controller_plugin.dir/src/gazebo_gimbal_controller_plugin.cpp.o
[51/72] Building CXX object CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o
[52/72] Linking CXX shared library librotors_gazebo_multirotor_base_plugin.so
FAILED: /usr/bin/c++   -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -Drotors_gazebo_mavlink_interface_EXPORTS -I/usr/include/opencv -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/akshay/src/Firmware/Tools/sitl_gazebo/include -I/home/akshay/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gazebo-9/gazebo/msgs -I/usr/include/eigen3 -I. -I/usr/include/eigen3/eigen3 -I/usr/local/include/OGRE -I/usr/local/include/OGRE/Paging -I/home/akshay/src/Firmware/mavlink/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -std=c++11 -std=c++11 -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++14 -MMD -MT CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o -MF CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o.d -o CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o -c /home/akshay/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp
/home/akshay/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘virtual void gazebo::GazeboMavlinkInterface::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/home/akshay/src/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:279:23: error: ‘class gazebo::physics::World’ has no member named ‘GetName’
   model_param(world_->GetName(), model_->GetName(), "mavlink_udp_port", mavlink_udp_port_);
                       ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1
```

I had previously followed the same method and that had worked correctly. However, I tried to start over and encountered this error.

Could you advise on how to proceed? Thanks.

TSC21, 2018-05-31:
What Gazebo version are you using?

akshaykvnit, 2018-05-31:
@TSC21 I am using Gazebo 9.0.0

TSC21, 2018-05-31:
Ok this should need a fix on `sitl_gazebo` to support Gazebo 9 aswell. This function was introduced in the latest commits for `sitl_gazebo` so this needs a compatibility fix (current CI builds over version 7). I will push a fix and let you know later today.

TSC21, 2018-05-31:
In the mean time, for it to work, regress the commit tag of `sitl_gazebo` to a previous commit (4 tags bellow should be enough).

TSC21, 2018-05-31:
The fix should include this: http://answers.gazebosim.org/question/16388/getname-has-not-been-declared-gazebo-9/. So `GetName()` should become `Name()` for Gazebo 9. You can also try to change it on the source code, but a fix will be pushed later today.

akshaykvnit, 2018-05-31:
Thank you for helping.

TSC21, 2018-06-01:
@akshaykvnit fix here: https://github.com/PX4/sitl_gazebo/pull/210

closed 2018-06-01 by LorenzMeier