PX4/PX4-Autopilot #9577: timesync error between px4 and mavros [simulation]

alexcherpi, 2018-05-31:
Hi,
When I run the two following commands in two different terminals the simulation works without errors.
1. no_sim=1 make posix_sitl_default gazebo (this runs px4)
2. roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/sitl_gazebo/worlds/iris.world

But when I run this third command in a third terminal, I get "ERROR [mavlink] [timesync] Hard setting offset." in the terminal running px4.
3. roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
In the mavros terminal I get this error: "[ WARN] [1527786373.403793887, 214.796000000]: TM: Clock skew detected (-0.013114400 s). Hard syncing clocks."

I'm running Ubuntu 16.04, with rosdistro: kinetic, with px4 v1.7.3.

What should I do ?

Here is the diagnostic
`header: 
  seq: 17
  stamp: 
    secs: 19
    nsecs: 442000000
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Received packets:"
        value: "5867"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "228"
      - 
        key: "Tx sequence number:"
        value: "248"
      - 
        key: "Rx total bytes:"
        value: "250190"
      - 
        key: "Tx total bytes:"
        value: "6187"
      - 
        key: "Rx speed:"
        value: "37359.000000"
      - 
        key: "Tx speed:"
        value: "521.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "3"
      - 
[... 305 lines cut ...]
# px4flow
px4flow:
  frame_id: "px4flow"
  ranger_fov: !degrees 6.8  # 6.8 degreens at 5 meters, 31 degrees at 1 meter
  ranger_min_range: 0.3     # meters
  ranger_max_range: 5.0     # meters

# vision_pose_estimate
vision_pose:
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "map"
    child_frame_id: "vision_estimate"
    rate_limit: 10.0

# vision_speed_estimate
vision_speed:
  listen_twist: true    # enable listen to twist topic, else listen to vec3d topic
  twist_cov: true       # enable listen to twist with covariance topic

# vibration
vibration:
  frame_id: "base_link"

# vim:set ts=2 sw=2 et:`

lamping7, 2018-06-03:
This error is normally a result of poor computer performance. Secondly, why are you launching all of these separatly instead of using the launch file px4 comes with to do all of this?

alexcherpi, 2018-06-19:
Thanks lamping7,
It should indeed be a result of poor computer performances. Nevertheless the good point is that everything is working and communicating (gazebo, ros, px4) except that I get in addition these warnings.
I am launching separately the gazebo ros and px4 processes in order not to relaunch everything if I just did some modifications in px4. And thus I only re-run the px4 process (keeping gazebo opened, indeed gazebo is quite slow to open on my computer).

mhkabir, 2018-06-19:
If you update to latest mavros and latest PX4, you might not get these warnings, since the timesync was improved to work on worse links.

alexcherpi, 2018-06-19:
I am working with the latest mavros but with PX4 v1.7.3
When was the commit to remove these warnings ?

mhkabir, 2018-06-19:
This is not from latest mavros : 
```
"[ WARN] [1527786373.403793887, 214.796000000]: TM: Clock skew detected (-0.013114400 s). Hard syncing clocks."
```

PX4 timesync : https://github.com/PX4/Firmware/pull/9365
Mavros timesync : https://github.com/mavlink/mavros/pull/1042

sunnyroad, 2018-08-02:
Hi,
would you help such issue I met on PX4?

If HEARTBEAT got before, and timesync one time success, everything is OK.
But if I manually change the linux systime  on my local system, then everything block suddenly after once HEARTBEAT received.
I had tried the lastest Mavros timesync : mavlink/mavros#1042, it do not work, same result.

closed 2019-01-22 by mhkabir