PX4/PX4-Autopilot #9578: arming and takeoff with mavros !

zubair123khan, 2018-05-31:
Hello

I am successfully able to launch mavros_posix_sitl.launch, but i am unaware of using mavros command like arming and takeoff. However i tried using rosservice call /mavros/cmd arming and set it to true but its returning me FALSE.
I don't think mavros  launch needs anything else for me to use service call for arming and takeoff.

please guide me on how can i arm and takeoff my iris in gazebo.

Thanks

TSC21, 2018-05-31:
`rosrun mavros mavsafety arm`
`rosrun mavros mavcmd takeoff 0 0 0 0 5` (being 5 the altitude)

closed 2018-05-31 by TSC21

zubair123khan, 2018-05-31:
Hi 
Thanks for the comment, although I tried these rosrun and it's showing up request failed

TSC21, 2018-05-31:
So you are probably missing something more fundamental here. Also it's not clear what's your current framework (meaning PX4 Firmware version, etc). Have you tried to launch the SITL simulation without ROS, by issuing `make posix_sitl_default gazebo` and after it opens (and gets a position lock), issue `commander takeoff` on the same window you launched the previous commad?

TSC21, 2018-05-31:
Please also mark here:
1. your ROS and MAVROS version;
2. you Gazebo version;
Also, try launching QGC and check for the data coming from the vehicle and if there are any log error msgs being reported

zubair123khan, 2018-05-31:
Did try it , commander take off without ROS is showing no position lock.. I waited for much longer..

TSC21, 2018-05-31:
> Did try it , commander take off without ROS is showing no position lock.. I waited for much longer..

OK are you able to show us a video/screencapture of the procedure you take to launch the simulation and then the behavior you see on the simulator? Are you using the latest upstream master of PX4 or a recent stable release?

zubair123khan, 2018-05-31:
This is my console output now without ROS 

-- http://pastebin.com/Hqx4sbxs

zubair123khan, 2018-05-31:
Here is my screen shot with mavros posix_sitl.launch 

-- http://pasteboard.co/HnN6Yih.png