PX4/PX4-Autopilot #9582: Failed loading mixer: /etc/mixers/quad_x.main.mix

stmoon, 2018-06-01:
Hi,

When I reboot quad copter using MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN, ARM didn't work with safety button off sometimes.
In this case I should power off and on again to work well.
 
And I found that [init] Failed loading mixer: /etc/mixers/quad_x.main.mix
Why does the mixer not work?

I have used v.1.7.4beta firmware.

Thanks

```
ERROR: Could not start: fmu mode_pwm 
ERROR  [init] Failed appending mixer: /etc/mixers/pass.aux.mix 
ERROR: Could not start: fmu mode_pwm 
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix 
ERROR  [init] Failed appending mixer: /etc/mixers/pass.aux.mix 
ERROR: Could not start: fmu mode_pwm 
PX4IO update failed 
PX4IO not found 
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix 
ERROR  [init] Failed appending mixer: /etc/mixers/pass.aux.mix 
ERROR: Could not start: fmu mode_pwm 
PX4IO update failed 
PX4IO not found 
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix 
ERROR  [init] Failed appending mixer: /etc/mixers/pass.aux.mix 
ERROR: Could not start: fmu mode_pwm 
```

ChrisHughes, 2018-06-03:
Same issue here, Pixfalcon and 1.7.4beta.

stmoon, 2018-06-03:
In my case, pixhawk 2.1 and same firmware

stmoon, 2018-06-04:
Could you please let me know the reason or some solution to fix it? 

Thanks.

LorenzMeier, 2018-06-17:
There is not nearly enough context to solve this. Please provide a complete writeup what you were trying to do, which configuration you were using and what you did before the reboot (including complete shell output).

This issue will auto-close in 30 days without any further input. There seems nothing to be wrong with the core system as far as we can tell with the information provided.

stmoon, 2018-06-17:
Thanks you @LorenzMeier ,

I just rebooted the PX4(pixhawk 2.1) using VEHICLE_CMD_PREFLIGHT_REBOOT_SHUTDOWN with param 1 using mavlink.
I have used firmware v1.7.4

I think that this problem happens once per 10 times try. and I found the bootlog.txt was written like this
```
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix 
```
and I found the error happen in the mixer in rc.interface.

```
if mixer load ${OUTPUT_DEV} ${MIXER_FILE}
```
Once the reboot is failed, next time the mixer does not working any more.
To solve this, I just should connect battery again for cold booting.

Thanks in advance

stmoon, 2018-06-21:
Hi @LorenzMeier 

Finally, I found the error when running "mixer load"

```
WARN  [px4io] fmu sent: "0
S: 3 5 10000 10000 0 -10000 10000
M: 1
O: 10000 10000 0 -"
px4io mixer send error -1
ERROR [mixer] failed to load mixers from /etc/mixers/quad_x.main.mix
ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix, ret:0
```

It is cause by the code in the mixer
```
	/* Pass the buffer to the device */
	int ret = px4_ioctl(dev, MIXERIOCLOADBUF, (unsigned long)buf);

```

Could you explain me why sometimes the warm rebooting is not working?

Thanks in advance

Best regards,
SungTae Moon

stmoon, 2018-06-24:
Is there someone to help me?

v01d, 2018-07-11:
I'm trying HIL and I found this on the bootlog, wonder if it is related since I get five low frequency tones on boot and I also cannot get hil actuator controls to be non-zero

stmoon, 2018-07-11:
Hi @v01d 

maybe yes,..  however I didn't find any solution until now. :-(

sunnyroad, 2019-06-06:
I also meet same issue, when reboot with mavlink cmd PREFLIGHT_REBOOT_SHUTDOWN, occasionally it do not work, and then mixer load fail, then same such reboot not work  any more, only power off-on could solve it.
[image]
[image]

Anyone can help this?

Best regards.

julianoes, 2019-06-07:
@sunnyroad on what Pixhawk is this? And does this happen when you reboot when armed or disarmed?

sunnyroad, 2019-06-27:
> @sunnyroad on what Pixhawk is this? And does this happen when you reboot when armed or disarmed?

Pixhawk2,  actually issue happen when reboot FCU via command "PREFLIGHT_REBOOT_SHUTDOWN", so after that, when we send armed command, it also respond armed success but no pwm output on each FCU's PWM_OUT IO(always zero output).

sunnyroad, 2019-09-04:
Hi, julianoes
   do you have any findings or fixs of this issue?

julianoes, 2019-09-04:
Does this issue still exist in v1.9?

sunnyroad, 2019-10-21:
> Does this issue still exist in v1.9?

Hi, we do not use v1.9,  is any related fix or enhances for this issue related after v1.8.2?

julianoes, 2019-10-22:
I don't think anyone really has the time to fix issues from before v1.9. If you find a way to reproduce it on v1.10, re-open this issue.

Or if you find a fix, we'd be happy to see a pull request, thanks!

closed 2019-10-22 by julianoes