PX4/PX4-Autopilot #9583: MC EKF2_HGT_MODE Set to LIDAR is not usable on non Flat Terrain

tops4u, 2018-06-02:
**Describe the bug**
This is not really a bug but the current implementation is IMHO not usable in the Field. 
I changed EKF2_HGT_MODE to Rangefinder as my Hexa features a Lightware SF20 (up to 100m). I'm searching for fawn and sometimes the terrain is not flat. In order to keep a constant Altitude to the Ground - which is needed in order to have all the ground covered in the Survey - I wanted to use the LIDAR. This basically and theoretically works. BUT: Manytimes there are Bushes and/or Trees in or on the edges of the Field. This causes the LIDAR to have Jump in altitude Measurement. 

Illustration:
Altitude Step on the LIDAR due to Trees that grow partially over the field and can not always be avoided:
[image]

Real Altitude:
[image]

Actuator Jumps due to Altitude Step:
[image]

Gyro Measurement due to High Frequent change in Trust and Trust going to min (approx 1050ms):
[image]

**Expected behavior**
Possibility to define a Max Slope that will trigger an altitude change (for instance in the above Situation a sudden 20m jump should then just fallback to Baro. For short Measurement Errors the HGT Checkbit is set correctly, it would be nice to be able to define a maximum Terrain Angle or change rate the would then knock out the LIDAR. 

Following https://docs.px4.io/en/sensor/rangefinders.html all the EKF2_RNG_* are only for Low Range US/Lidar to keep short distances and aid when landing, while EKF2_HGT* does not offer any advanced configuration Options. If I'm just using it wrong, please advise how to prevent the above situation. 

Link to the full Log: https://logs.px4.io/plot_app?log=b72cb675-2f25-4f95-a2e6-5ba7da4d3fa0

tops4u, 2018-06-04:
Add On: There is also MPC_ALT_MODE Parameter. 
But no real description who this works and some sort of a warning: "_Altitude control mode, note mode 1 only tested with LPE_"

priseborough, 2018-06-04:
Terrain following should be done by the position controller, not the estimator. Meeting this use case will require modification to mc_pos_control so that it uses local_position.dist_bottom robustly.  It is likely retuning of the single state estimator in ekf2 that estimates the offset of the terrain relative to the local earth frame datum will be required so that it is less reactive to range finder changes.

Pull request https://github.com/PX4/Firmware/pull/9509 introduces support for terrain following with mc_pos_control when using ekf2, but is unlikely to meet your use case without modification.

tops4u, 2018-06-04:
@priseborough Does MPC_ALT_MODE have any function with EKF2? Or will this only work when #9509 or another PR is in place? Would MPC_ALT_MODE = 1 then take any Range Device that it can find (US, LIDAR, Radar)?

tops4u, 2018-06-04:
Just to be sure, if a Range Finder of any sort is available EKF2 would already now make use of this data (without setting MPC_ALT_MODE)??

priseborough, 2018-06-04:
yes, the ekf2 will use range data provided it is not outside the range specified by the uORB topic  range_finder.min_distance and range_finder.max distance_distance messages. If on ground and outside the valid range specified by the uORB data, then the EKF2_MIN_RNG value will be used.

priseborough, 2018-06-07:
Progressively halve the  terrain process noise EKF2_TERR_NOISE from the default value of 5.0 m/s. That will make the terrain height estimator less reactive to range finder noise.

tops4u, 2018-06-13:
@priseborough  Thank you very much for your help and explanation. I'll close this and try to figure proper Params for TERR_NOISE for my use case.

closed 2018-06-13 by tops4u