PX4/PX4-Autopilot #9584: BUG: AUX Outputs swinging wildly while RC System is off

tops4u, 2018-06-02:
**Describe the bug**
While working on the bench, the Transmitter is sometimes off. I noticed that despite all Main PWM Signals are tied to Disarmed PWM. But the AUX Outputs are not. When I power on my UAV while the TX is off, the FMU does not recognize the RX System, as I'm using Futaba SBus and there is apparently no signal on the bus if the system has not been active at least once. In the Radio Panel of QGC no Transmit-Channel can be seen (lower righthand side). 
I noticed that my Flashlights are going mad when I perform the above steps. I'm not 100% sure how the Mapping is done for mRo 2.1 but I Suspect, that ServoChannels 13-16 are be mapped to the AUX Outputs. 

While the described above Situation the following Curve of Servo RAW 13 has been analyzed (via 433MHz Telemetry):

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Using a SBus System while the TX is Powered Off
2. Power the UAV (you may also Power the RC RX). 
3. Check AUX Outputs for activity

**Expected behavior**
While there is no RC System detected all Servo Outputs should stay put. 

**Drone (please complete the following information):**
- mRo X-2.1
- Futaba R7008SB (Channel 7 Mapped to AUX1 / Channel 8 to AUX2). 
- PX4 1.8 BETA

bkueng, 2018-06-04:
I'm not sure what QGC shows, but port0 indicates, that these are all main outputs. And at the moment, the Firmware fills in only channels 1-8 and sets the others to 0 (https://github.com/PX4/Firmware/pull/9586), so I'm not sure what QGC shows as servo 13.

Can you do the following:
-Provide the output of `fmu status` and `px4io status`
- The AUX outputs that swing are CH1 and CH2 on https://docs.px4.io/en/flight_controller/mro_x2.1.html#wiring-guide?
- Can you record a log while the outputs are swining? -> `logger on`, `logger off`

tops4u, 2018-06-04:
@bkueng I noticed myself meanwhile that Servo does not map to AUX. I can attach an Oscilloscope to check what really happens, and post the result here. Yes at least AUX 1 is not stable as it triggers my strobes when RC System is off. As soon as I switch on the TX this, does not happen.

tops4u, 2018-06-06:
Could not reproduce. I'm closing this here.