PX4/PX4-Autopilot #9590: Error with MAV_CMD_LOGGING_START and STOP

pfrck, 2018-06-04:
So I'm getting this whenever I try to run [mavlink_ulog_streaming.py](https://github.com/PX4/Firmware/blob/master/Tools/mavlink_ulog_streaming.py): 

```Output file name: .\2018-06-04_17-20-17.ulg
Connecting to MAVLINK...
Starting log...
Traceback (most recent call last):
  File "Y:/testbench/2.1.3/mavlink_ulog_streaming.py", line 267, in <module>
    main()
  File "Y:/testbench/2.1.3/mavlink_ulog_streaming.py", line 255, in main
    mav_log_streaming.start_log()
  File "Y:/testbench/2.1.3/mavlink_ulog_streaming.py", line 59, in start_log
    mavutil.mavlink.MAV_CMD_LOGGING_START, 0,
AttributeError: 'module' object has no attribute 'MAV_CMD_LOGGING_START'
```

It looks like I don't have the related command specified in mavlink or mavutil. I didn't find anything similar to this and my bad if I didn't search enough but I could use some help there.

bkueng, 2018-06-04:
Most likely your `pymavlink` is not up to date. Can you try updating it?

pfrck, 2018-06-05:
That was exactly it. I had the 2.0.6 version and upgraded it to 2.2.8 and it works now. Thanks!

pfrck, 2018-06-05:
Well thought it was resolved by upgrading pymavlink to version 2.2.8 but now I'm getting this:
```
COMMAND_ACK {command : 2510, result : 3}
```
which means that the command 2510 MAV_CMD_LOGGING_START is unsupported (result: 3).

ajaiRevision, 2018-08-03:
@bkueng and @pfrck ,

Hi,
@pfrck , did you resolve the issue?

I am also facing the same issue mentioned first (no attribute called MAV_CMD_LOGGING_START). I did try fresh install of pymavlink (pip install pymavlink). However, my error remains the same. I did not get the 2510 error.

What else could be the problem? Any ideas?