PX4/PX4-Autopilot #9591: Convert racer joystick rates to PX4 acro rates

MaEtUgR, 2018-06-04:
I think it would be useful to support racers who come from a different FC to test PX4 and let them convert their existing rates in cleanflight betaflight fork parameters to our better tunable solution.

Like I promised here
https://github.com/PX4/Firmware/pull/9358#issuecomment-385896362
https://github.com/PX4/Firmware/pull/9358#issuecomment-394408435
I provide the calculation script and it would probably be easiest to just run it on a website but best to include the functionality in QGC.

dagar, 2018-06-04:
That's a great idea.

MaEtUgR, 2018-06-06:
I realized existing FCs also have differences and I found an existing easy to use open source tool which already converts between them and visualizes the result: https://apocolipse.github.io/RotorPirates/

I could add PX4 rates and we can link to it in the documentation as soon as I understand where all the calculations are done: https://github.com/apocolipse/RotorPirates/issues/4 That's more future proof because it can directly be maintained if anything changes for any FC in the future.

MaEtUgR, 2019-01-24:
The maintainer of the value converter repo finally answered me now and I'll add the PX4 stick mapping to https://apocolipse.github.io/RotorPirates/

MaEtUgR, 2019-01-26:
PR for this now here: https://github.com/apocolipse/RotorPirates/pull/7

LorenzMeier, 2019-01-26:
Very cool!

MaEtUgR, 2019-06-25:
Recently I looked into the rate calculation of betaflight and it's in theory slightly different because they use an x^4 term as expo instead of x^3 like we do. But it's in practise totally different because they still use these totally unintuitive configuration parameters that all depend on each other...

I think a first step is to write both formulas down clearly and then figure out an approximate coefficient transformation.

MaEtUgR, 2019-06-25:
**PX4**
https://github.com/PX4/Firmware/blob/6d9bf18ca6e1ef4c3a721832398cae72385bf360/src/lib/mathlib/math/Functions.hpp#L61-L95
`y = RateMax * (1 - Expo) * x + Expo * x * x * x * (1 - Superexpo) / (1 - abs(x) * Superexpo)`

**Betaflight**
Here's the betaflight rate formula that we should be compatible to in some documented way: https://github.com/betaflight/betaflight/blob/ecb67fcc097d284db706149bdeefaa40b17ce921/src/main/fc/rc.c#L120-L138

The extracted formula without corner case handling is:
`y = 200 * rcRate * (x * x^3 * Expo + x * (1 - Expo)) * 1 / (1 - (abs(x) * (SuperRate / 100.0f)));`

So we need to convert the scaling and the expo to approximate x^4.

junwoo091400, 2022-07-27:
This is quite cool! And maybe worth keeping it open :raised_hands:

hamishwillee, 2025-03-27:
@MaEtUgR I've removed the Documentation label. If we ever do something with this, let me know.