PX4/PX4-Autopilot #9594: CMake warnings are thrown while building PX4 firmware in Debian 9.3

Ankur1014, 2018-06-05:
**Describe the bug**
Building PX4 Firmware is throwing warnings about CMake in Debian 9.3

**To Reproduce**
I installed the toolchain and downloaded the PX4 Firmware version 1.6.5 in Debian 9.3. While building the firmware using _make px4fmu-v2_default_ command I am seeing the following warnings in Build.

-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5
-- CONFIG: nuttx-px4fmu-v2-default
**CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  /usr/share/cmake-3.7/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:275 (project)**

**CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  /usr/share/cmake-3.7/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:275 (project)**

**CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:31 (cmake_force_c_compiler)
  build_px4fmu-v2_default/CMakeFiles/3.7.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:275 (project)**

**CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:37 (cmake_force_cxx_compiler)
  build_px4fmu-v2_default/CMakeFiles/3.7.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:275 (project)**

-- The ASM compiler identification is GNU
-- Found assembler: /home/abose/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.13") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- Adding ROMFS on px4fmu-v2
-- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
-- Configuring done
-- Generating done
-- Build files have been written to: /mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default
[72/573] Building NuttX for px4io-v2 with nsh
**make[1]: Warning: File '/mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default/px4io-v2/NuttX/nuttx/.config' has modification time 0.025 s in the future
make[2]: Warning: File '/mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default/px4io-v2/NuttX/nuttx/.config' has modification time 0.0076 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File '/mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default/px4io-v2/NuttX/nuttx/nuttx-export/Make.defs' has modification time 0.2 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
[73/573] Building NuttX for px4fmu-v2 with nsh
make[1]: Warning: File '/mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default/px4fmu-v2/NuttX/nuttx/.config' has modification time 0.0029 s in the future
make[2]: Warning: File '/mathworks/home/abose/px4_165/src/Firmware/build_px4fmu-v2_default/px4fmu-v2/NuttX/nuttx/nuttx-export/Make.defs' has modification time 0.3 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: warning:  Clock skew detected.  Your build may be incomplete.**
[573/573] Generating px4fmu-v2_default.px4, firmware_nuttx.bin

**Expected behavior**
A build without warnings. I never saw these warnings in Ubuntu 16.04

**Additional context**
Are these warnings okay to have ? I did not yet test the final build

dagar, 2018-06-06:
Can you repeat with PX4/Firmware master?

Ankur1014, 2018-06-22:
Hi ,
I repeated this step with PX4 Firmware master. It is still reproducible

mcsauder, 2018-07-17:
Hi @Ankur1014 , I've seen an error like this before but not exactly yours...

Please run a `make clean`, (even perhaps reboot your computer), and make sure you have an internet connection and that your system clock matches approximately real time before you build. With my machine this happens because I have a dual boot Windows/Linux machine and each OS fights for control of the system clock.  When I boot to Windows then back to Linux it resets my system clock to GMT, then all of my builds are messed up until 6 hours later or a `make clean`.

I hope this helps.

-Mark

closed 2019-07-09 by stale[bot]