PX4/PX4-Autopilot #9595: Pixhawk 4, 1.8.0 beta 2 crash

r0gelion, 2018-06-05:
Using Pixhawk 4 autopilot with 1.8.0 beta-2 (ab279d5f) firmware, testing mc02 test card which is a full autonomous mission.

The vehicle took off and started to follow the mission, after waypoint 4 the vehicle fell out of the sky.
[image]

The mission was started from QGC and the RC throttle was at the minimum, the log shows that after 10:09 the vehicle changed flight mode to stabilize and fall due to the low Throttle.

The question is why the vehicle exited mission and changed to Stabilize?

Should the vehicle hover, change to altitude or any other option instead of changing to Stabilize when the mission is aborted?

Here you can find the log
https://review.px4.io/plot_app?log=51293c49-2708-4356-b5f2-11c4ca820a3a

and the [tlog](https://github.com/PX4/Firmware/files/2071166/2018-06-04.10-02-44.zip)

Any question please let me know.

thomasgubler, 2018-06-06:
TODO from dev call: analyze why it exited mission. Crash was because stick was at low throttle when in switched to  stabilized

LorenzMeier, 2018-06-06:
I had a pass this morning and couldn't figure out what happened in the first 15 minutes.

bkueng, 2018-06-07:
At the time of the incident, the flight mode switches to RTL and then immediately to stabilized. At the same time the kill switch goes on and immediately off again.
RC messages do not reflect that, but we don't log at full rate. Only pitch and yaw sticks change minimally.
@r0gelion do you know if the pilot had his hands close to the RC sticks when the switch out of mission happened?

It might be an RC glitch, however that does not explain why the system did not switch back to mission mode after the glitch.

tops4u, 2018-06-07:
I don't want to reopen my case (https://github.com/PX4/Firmware/issues/8263) but remind you that there have been similar reports about Kill-Switch Toggles at a speed that could not be triggered so fast manually by the pilot. Also there is different HW / FW involved. Maybe this is RC System related. I was using Futaba SBus (T-FHSS RX).

r0gelion, 2018-06-07:
@bkueng The pilot says the RC wasn't close to his hands, it was in front of him on a table, but with a distance that prevents unnoticed manual input.

LorenzMeier, 2018-06-08:
Not a crash, but suspicious: https://review.px4.io/plot_app?log=02711563-1bd4-40ee-8107-16fa14017b9e

dagar, 2018-06-13:
There's not a lot to go on here, but the log does catch occasional blips at other times.

[image]

Adding a small hysteresis to the switch changes wouldn't hurt, although I can't claim that would have prevented this.

LorenzMeier, 2018-06-15:
It can't be the publication, I checked. But the plot you saw is harmless - there are only 8 initialised channels. I fixed the logging artefact here:
https://github.com/PX4/Firmware/pull/9690

LorenzMeier, 2018-06-15:
I think this:
https://github.com/PX4/Firmware/blob/20aabd35665e53dedde67064e895127f004da914/src/modules/commander/commander.cpp#L2995

Is the culprit, triggered by a combination of low battery state and RC glitch leading to RTL. Will look more closely tomorrow.