PX4/PX4-Autopilot #9601: Fixed-wing control bug: Stuttering rudder even in manual mode

philipoe, 2018-06-05:
**Describe the bug**
For fixed-wing aircraft, even in manual control mode, the rudder is stuttering significantly as soon as a constant aileron input is given via a remote controller. This was tested on 3 different platforms with a PX4FMUv3 with the current master, but the bug has existed for quite some time. This is a low-level control bug that should be removed ASAP. It is linked to the FW_RLL_TO_YAW_FF parameter, i.e. if that parameter is zero you will not see the issue.

**To Reproduce**
- Set FW_RLL_TO_YAW_FF > 0. In the screenshot below. we use FW_RLL_TO_YAW_FF=0.2 .
- go to manual mode and give a constant aileron input
- The rudder stutters (although you might not be able to see that due to the mechanical setup of your platform). The actuator_control_target0 group shows that control0 (aileron) is constant, then control3 (rudder) is jumping.
[image]

**Root cause**
The issue is that [this line](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L828) increments the rudder output every time that it is called, however, the actual manual control input (i.e. the base value which is incremented) is only reset when the `manual_control_setpoint topic` is updated. So if the controller runs at say 100Hz, but the manual_control_setpoint topic is only updated at 50Hz, then the rudder output will be incremented ca. 2 times by FW_RLL_TO_YAW_FF before it is reset to the actual control input. This is what you see in the screenshot above. Now the actual check whether manual_control_setpoint was updated is done [here](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L270), and depending on that the _actuators variable is assigned a new value or not [here](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L331). 

I tested that just assigning replacing [line 270](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L270
) by `if (_vcontrol_mode.flag_control_manual_enabled)`, i.e. ALWAYS assigning the _actuators variables even if the respective manual control inputs were not updated, works. The final question is: Does anything speak against just always updating these variables? That would solve the problem.

LorenzMeier, 2018-06-05:
Off the bat I’d say always updating is fine.

philipoe, 2018-06-06:
Fixed in ba2cf5fa9a20a1577f9a9153df4f793f9e49c8f6, closing.

closed 2018-06-06 by philipoe