PX4/PX4-Autopilot #9602: failsafe after takeoff shouldn't climb first

julianoes, 2018-06-05:
In SITL, when I do a normal `commander takeoff` test, the drone first goes to the takeoff altitude, then it goes into failsafe because there is neither RC nor a data link. This means it will rise to a default altitude of 5 meters before descending, loiter, and finally landing.

The fact that it rises first in failsafe when it is right over the takeoff spot is not intuitive in my opinion. Better would be if it just descended and landed again.

```
pxh> commander takeoff
pxh> INFO  [commander] Takeoff detected
WARN  [commander_tests] Failsafe enabled: no datalink
INFO  [navigator] RTL HOME activated
INFO  [navigator] RTL: climb to 493 m (5 m above home)
INFO  [navigator] RTL: return at 493 m (5 m above home)
INFO  [navigator] RTL: descend to 493 m (5 m above home)
INFO  [navigator] RTL: loiter 5.0s
WARN  [navigator] RTL: land at home
INFO  [commander] Landing detected
```

lamping7, 2018-06-05:
This conversation seems related: https://github.com/PX4/Firmware/pull/8738

Antiheavy, 2019-01-22:
Fixed Wing might need different handling for this.

bresch, 2019-03-25:
Someone reported that the vehicle also goes up to reach RTL altitude even if the drone is already above home, ready to land. Not sure if this is true, but we have to remember to check those cases as well.

bresch, 2019-06-25:
We need to implement a sort RTL cone to solve that. I even had a case where the drone was on the ground and just not disarmed when the battery failsafe RTL was triggered.

FYI @MaEtUgR

bresch, 2019-09-23:
Solved by #12382

MaEtUgR, 2019-10-21:
Not really. By default it's not enabled and it still climbs to a certain minimum altitude even when enabled.

hamishwillee, 2020-06-05:
Is this fixed?

junwoo091400, 2022-07-27:
Now the half cone angle for RTL is set to 45 degrees by default. So it should work:

https://github.com/PX4/PX4-Autopilot/blob/f7ff0a996102ed020d0cbf53b530b6ba5a2cd721/src/modules/navigator/rtl_params.c#L138

junwoo091400, 2022-07-27:
But, I do notice that the logic for caculating this half cone RTL altitude is only for rotary wing (shouldn't it be only for fixed wings?!!), will create PR for that.

junwoo091400, 2022-07-27:
Hum this is a bit weird. Here: https://github.com/PX4/PX4-Autopilot/pull/15216, Cone logic isn't used for fixed-wings. But why?

julianoes, 2022-07-28:
@junwoo091400 this problem might indeed be solved now. It was not actually related to fixedwing at all. It's still wrong/stupid how it goes into failsafe after arming instead of preventing arming but that's another story.

closed 2022-07-28 by julianoes