PX4/PX4-Autopilot #9606: Mission starts at closest waypoint

potaito, 2018-06-06:
Behaviour introduced with PR https://github.com/PX4/Firmware/pull/8749.

Opening an issue on @dagar's request.

Description:
1. Takeoff and leave the drone in position hold.
[image]

2. Create waypoint mission where some intermediate waypoint is close to the current location.
[image]

3. Start mission. The vehicle will jump to the closest waypoint and commence the mission from there, skipping all waypoints before that.
[image]

The changes responsible for this are these two blocks in `mission::on_activation()` and `mission::on_active()`, setting the current waypoint to the closest mission item.

https://github.com/acfloria/Firmware/blob/a70cd4e54567ae774ada1d0197797890c16e717f/src/modules/navigator/mission.cpp#L155-L158

https://github.com/acfloria/Firmware/blob/a70cd4e54567ae774ada1d0197797890c16e717f/src/modules/navigator/mission.cpp#L198-L206

Tested on current master (ca5e22583f9dd81852856a9169523d8135b81db6) with SITL `make posix_sitl_default jmavsim` and default parameters.

Antiheavy, 2018-06-07:
I agree that for 1.8.0 release this behavior should probably be changed backed to the way it was before for all auto modes except the new RTL_TYPE=2.  However, I think there should be a bigger discussion about if the default mission item handling when uploading a new mission could be improved after 1.8.0.  PX4 resets the mission index to 1 when a new mission is uploaded; this is not how most professional systems have traditionally worked and many users find it frustrating/confusing.  I'd be happy to open a separate "discussion" issue about this for future consideration if people are willing to discuss/debate.

LorenzMeier, 2018-06-09:
@acfloria @dagar Do you have cycles this weekend to make the change? I really would like to get v1.8 out of the door.

Thanks!

LorenzMeier, 2018-06-12:
@dagar Any news on this?

acfloria, 2018-06-12:
I am currently working on it.

commit 4697bf42 (2018-06-12): Navigator Fix: Do not set the closest mission item in normal mission mode (#9646)

 - fixes #9606
closed 2018-06-12 by dagar