PX4/PX4-Autopilot #9609: Set PWM_MIN for DJI F450 frame

LorenzMeier, 2018-06-06:
**Describe the bug**
F450 builds seem to stop props if PWM_MIN is set to 1075. We might need to set them to 1175 or an alike slightly higher setting

**To Reproduce**
Steps to reproduce the behavior:
1. Switch the drone on and arm it without propellers attached
2. Tilt it 60 degrees

**Expected behavior**
The propellers should not stop

**Drone (please complete the following information):**
- DJI F450 frame

**Additional context**
This can lead to crashes if it happens in-air.

bkueng, 2018-06-06:
This largely depends on the ESCs. Are you refering to using the original DJI ESCs?
In my experience an ESC calibration is usually needed and should be part of a vehicle setup. Enabling one-shot makes a difference too.

dagar, 2018-06-06:
I've come across a lot of people selecting this in PX4 as a generic 450 sized frame. If this PX4 airframe is going to become more specific I suggest we also introduce a range of intentionally generic options.

thomasgubler, 2018-06-06:
Output from Dev call: DJI ESC don't do calibration

mcsauder, 2018-06-06:
Hi @LorenzMeier ,

I've dug into this issue a bit.  The configuration value invoked by `4011_dji_f450` sets `PWM_MIN 1230`, so something else is causing your issue.

When `4011_dji_f450` gets called in the startup scripts, it then calls `4001_quad_x` which in turn calls `rc.mc_defaults`.  `rc.mc_defaults` sets `PWM_MIN 1075`, (only if `AUTOCNF == yes`).  Once `rc.mc_defaults` completes, `4001_quad_x` completes, then `4011_dji_f450` sets `PWM_MIN 1230`, (again, only when AUTOCNF == yes).

Although the default value of PWM_MIN in the startup scripts is initialized at 1000, I think the cause of this issue might have been either not configuring the airframe correctly, or manually setting that value lower sometime after the airframe was set.

Let me know your thoughts.

-Mark

thomasgubler, 2018-06-13:
TODO: verify analysis of @mcsauder by setting f450 config and observe values

mcsauder, 2018-06-13:
I've verified that uploading the current master branch v1.8.0-rc0 firmware, if the F450 airframe is selected PWM_MIN is set to 1230.  If, however, the F450-sized quadrotor with CAN (4012_quad_x_can), PWM_MIN remains the default value of 1075.  I've also checked F350 and verified the F350 airframe sets PWM_MIN to 1230.

Should set PWM_MIN 1230 perhaps be added to the 4012_quad_x_can config file?

bkueng, 2018-06-13:
> Should set PWM_MIN 1230 perhaps be added to the 4012_quad_x_can config file?

From what I see uavcan does not use `PWM_MIN`.

Also I'm currently updating the MC PID tuning page and I specifically added `PWM_MIN` as a precondition: https://bkueng.gitbooks.io/px4-user-guide/content/en/advanced_config/pid_tuning_guide_multicopter.html#precondition

bkueng, 2018-06-15:
@LorenzMeier given that the default is set to 1230, and your initial reporter changed it to 1075, I don't think we need to change anything. What do you think?

LorenzMeier, 2018-06-15:
Yep. Let's close this!

closed 2018-06-15 by LorenzMeier