PX4/PX4-Autopilot #9616: gazebo fails to start on macOS due to protobuf exception

julianoes, 2018-06-07:
Currently, with latest master and after a `brew upgrade`, gzserver throws an exception on start:

```
Gazebo multi-robot simulator, version 8.4.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.35.93
[libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: Int32.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1394] CHECK failed: generated_database_->Add(encoded_file_descriptor, size):
libc++abi.dylib: terminating with uncaught exception of type google::protobuf::FatalException: CHECK failed: generated_database_->Add(encoded_file_descriptor, size):
/Users/julianoes/src/Firmware/Tools/sitl_run.sh: line 119:  7882 Abort trap: 6           gzserver --verbose ${src_path}/Tools/sitl_gazebo/worlds/${model}.world
SITL COMMAND: /Users/julianoes/src/Firmware/build/posix_sitl_default/px4 /Users/julianoes/src/Firmware /Users/julianoes/src/Firmware/posix-configs/SITL/init/ekf2/iris
data path: /Users/julianoes/src/Firmware
commands file: /Users/julianoes/src/Firmware/posix-configs/SITL/init/ekf2/iris
```

FYI @priseborough

bramsvs, 2018-06-09:
@julianoes Problem is this commit: https://github.com/PX4/sitl_gazebo/commit/de983cc09955cdb1e6e68b1cef5116cd4750f18f?diff=split it works when you comment out all the refs to `std_msgs`, see here: https://github.com/bramsvs/sitl_gazebo/commit/bf6d22af29354ab81f5d7427dc8ea2d634a417de

Didn't test it further and my knowledge about px4/protobuf is quite limited so we need more input on a proper fix. Anyone? :)

LorenzMeier, 2018-06-10:
Thanks! Fix is on master: https://github.com/PX4/Firmware/commit/3da459899a9b892c28752203e922834672641676

closed 2018-06-10 by LorenzMeier

dagar, 2018-06-10:
Do we need MacOS CI building gazebo?

TSC21, 2018-06-10:
@dagar working on it: https://github.com/PX4/sitl_gazebo/pull/217

LorenzMeier, 2018-06-10:
@TSC21 has a PR up for that, WIP.