PX4/PX4-Autopilot #9632: Enable SITL compile test on non-ROS systems

LorenzMeier, 2018-06-10:
**Describe the bug**
We are incurring instabilities on non-ROS enabled systems (what our downstream adopters are using) due to a lack of CI on the Gazebo integration

**To Reproduce**
Run `make posix gazebo`

**Expected behavior**
Should build

**Log Files and Screenshots**
It fails to load plugins because we are 'faking' a ROS environment and have namespace clashes with the Gazebo standard message namespace.

```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.4
[libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: Int32.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1394] CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
libc++abi.dylib: terminating with uncaught exception of type google::protobuf::FatalException: CHECK failed: generated_database_->Add(encoded_file_descriptor, size): 
/Users/lorenzmeier/src/Firmware/Tools/sitl_run.sh: line 119: 62036 Abort trap: 6           gzserver --verbose ${src_path}/Tools/sitl_gazebo/worlds/${model}.world
SITL COMMAND: /Users/lorenzmeier/src/Firmware/build/posix_sitl_default/px4 /Users/lorenzmeier/src/Firmware /Users/lorenzmeier/src/Firmware/posix-configs/SITL/init/ekf2/iris
data path: /Users/lorenzmeier/src/Firmware
commands file: /Users/lorenzmeier/src/Firmware/posix-configs/SITL/init/ekf2/iris
```

LorenzMeier, 2018-06-10:
@dagar We need SITL on one of the Mac OS build systems and validate the sitl_gazebo repo against it.

@TSC21 We need to stay away from the Gazebo or ROS message namespace for our own messages as that will create constant namespace clashes and hard-to-debug-or-maintain problems. The recent 'cleanup' was unfortunately introducing a lot of these potential issues and we have now already the first problem. It compiles, but it explodes on execution.

TSC21, 2018-06-10:
Yeah I will merge it back into one lib for custom proto msgs in Gazebo. This issue does not happen in Linux systems (tested on Ubuntu 16.04). So we definetely need CI on MacOS, since every change should be tested system-wise (I don't own a Mac for example, so I am not able to test everything on each platform).

TSC21, 2018-06-10:
@LorenzMeier I think the problem is more simple: can you please take https://github.com/PX4/sitl_gazebo/blob/master/CMakeLists.txt#L246 out and retest? I think the problem is `Int32` being added into two separate libs (a typo), nothing to do with Gazebo/ROS namespace clash.

closed 2018-06-10 by LorenzMeier

dagar, 2018-06-10:
Can we create a fresh issue describing what's needed?