PX4/PX4-Autopilot #9639: Snapdragon broken on master (11.06.2018)

RomanBapst, 2018-06-11:
**Describe the bug**
While trying to test #9482 noticed that master is broken for Snapdragon flight.
It seems that yet another missing symbol was introduced, see below

```
[08500/03]  02:17.372  HAP:4152:undefined  symbol _FNan (783) /libpx4.so  0303  symbol.c
```

RomanBapst, 2018-06-11:
Probably related to this
https://github.com/PX4/Matrix/commit/03a3e3ad46d21bb90e78f8cedd3526d24617df5f

RomanBapst, 2018-06-11:
Bisect results:
First bad commit: 5d066f95c72550626f7ee72b7bff930042485360
which is the matrix lib update

dagar, 2018-06-11:
Ok, that's an easy one to fix. We can use INFINITY from C instead of C++ limits.

For snapdragon CI we need
1. snapdragon docker containers that produce working binaries (likely separate for eagle and excelsior)
2. a snapdragon linux package that contains all the px4 pieces linux side + DSP shared libs
3. a snapdragon flight connected to a Jenkins slave via adb serial or over the network

At the moment 1 is the blocker.

closed 2018-06-12 by LorenzMeier