PX4/PX4-Autopilot #9640: [load_mon] logger low on FDs!

philipoe, 2018-06-11:
Issue:
On Pixhawk (mRo PX4FMUv2.4.6, but that should not matter) and the current master we use the logger app, but use a custom logger_topics.txt called e.g. [here](https://github.com/PX4/Firmware/blob/537451c2a09c18d76a0bdba975c59a0f7e171ce2/src/modules/logger/logger.cpp#L833) to load a couple of topics in addition to the ones loaded by default [here](https://github.com/PX4/Firmware/blob/master/src/modules/logger/logger.cpp#L604). This of course requires more file descriptors to subscribe to these topics. Therefore, our log files contain "low on FDs" warnings and the corresponding topics are _not_ logged:

> WARNING [load_mon] logger low on FDs! (1 FDs left)
[...]
[logger]orb_subscribe_multi distance_sensor failed (5)
[logger]orb_subscribe_multi telemetry_status failed (5)
[...]

Question/ How to solve:
I guess that the current number of file descriptors assigned to the logger app (and every other app as i understood?) [here](https://github.com/PX4/Firmware/blob/master/platforms/nuttx/nuttx-configs/px4fmu-v2/nsh/defconfig#L784) is just not enough to allow the user to add any meaningful amount of additional topics to be logged (we maybe add 5 additional topics). Question: Can this be increased (to something like 60 or 65) without influencing other apps too much? Or is there another way to just give more available FDs _only_ to the logger app?

dagar, 2018-06-11:
The short answer is we probably can't increase the number of file descriptors on fmu-v2 hardware, but we could consider it on v5 (pixhawk 4). It burns a lot of memory per nuttx task.

I actually have an old test branch that lets logger (and others) bypass file descriptors for uORB, but that's a much larger discussion.

As an immediate workaround we could review your entire logger list (builtin + logger_topics). There are likely messages that won't be missed.

philipoe, 2018-06-11:
> The short answer is we probably can't increase the number of file descriptors on fmu-v2 hardware, but we could consider it on v5 (pixhawk 4). It burns a lot of memory per nuttx task.

OK, good to know!

> As an immediate workaround we could review your entire logger list (builtin + logger_topics). There are likely messages that won't be missed.

Thanks, we already started that. Closing this issue for now.

closed 2018-06-11 by philipoe