PX4/PX4-Autopilot #9648: UAVCAN 1 and 2 on master (2018-06-12) broken

thomasgubler, 2018-06-12:
@DanielePettenuzzo opening this issue to track against 1.8. Please fill out and feel free to edit the title :)

**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

DanielePettenuzzo, 2018-06-12:
FYI @LorenzMeier @davids5 @thomasgubler

davids5, 2018-06-12:
@DanielePettenuzzo as mentioned here https://github.com/PX4/Firmware/issues/9534#issuecomment-395893390

>Also in case it is not mentioned anywhere, be advised that FW upgrade and dynamic node ID assignment ONLY will occur on the CAN1 bus. This is by design.

What this means is the FW server (for historical RAM size limitation) a separate task under uavcan on the FMU is that is started in the configuration phase and then stopped to conserver RAM. It is responsible for 1)  dynamic node ID allocation (think DHCP on can) 2) FW upgrade - using the brickproof bootloader on a UAVCAN node to update the fw on the node. 3) Parameter setting on the nodes.  This is where the hardfault was happening and why it is only related to CAN1.

DanielePettenuzzo, 2018-06-12:
UAVCAN FW server is not running on CAN2 therefore the function that causes the hardfault is never called.

thomasgubler, 2018-06-12:
Rough ETA I got from Daniele is noon CEST 2018-06-13

LorenzMeier, 2018-06-12:
@DanielePettenuzzo There is also the PR from David with the UAVCAN update which might address a couple of issues.

dagar, 2018-06-12:
See https://github.com/PX4/Firmware/pull/9650

davids5, 2018-06-12:
@DanielePettenuzzo - I think this was related to a race that happens because of the F7. I could have been speed, ordering or  having 2 CAN interfaces. The Fix is #9652. I found that send_uavcan was called before enque_uavcan_request on the F7. This was not the case on a AUAV X2.1. I am not sure if it related to having 2 can interfaces or we would have seen it on FMUv2 w/2MB

DanielePettenuzzo, 2018-06-12:
@davids5 thanks you. Today I also tried with Pixhawk 3 Pro (fmu-v4pro) that has 2 can interfaces and I didn't have this issue.. Instead with the Pixhack v5 (also fmu-v5) I have the same issues as on Pixhawk 4.

davids5, 2018-06-12:
@DanielePettenuzzo  For the record my assumption above about a race was wrong! The bug has been in the code forever. It was not a race. The dereferencing null on the F4 did not pitch the exception but does on the F7.

Please cross test the PR as well.

DanielePettenuzzo, 2018-06-12:
@davids5 will do one more test first thing tomorrow morning although this is the same change I did today and it was working fine.

davids5, 2018-06-12:
@DanielePettenuzzo - thank you! The value add will be testing on the new libuavcan that is now on master.

DanielePettenuzzo, 2018-06-12:
@davids5 perfect. will do.

closed 2018-06-12 by davids5