PX4/PX4-Autopilot #9667: CI mission test intermittent failures

dagar, 2018-06-13:
Recently the CI mission tests have started failing more regularly with PX4 appearing to exit almost immediately.

[image]

Examples
3 in this PR - http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/master_gps_fix/2/pipeline
2 in this ecl update test - http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/ecl_PR-458/3/pipeline
2 more - http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/master_gps_fix/1/pipeline

@lamping7 @bkueng FYI

lamping7, 2018-06-13:
The shutdown is caused by an assertion in the test. The test isn't getting messages from the ROS topics. If messages are not received 60 seconds after startup the test fails and quits. This normally takes less than 15 sec.

Something is possibly wrong with MAVROS or sitl_gazebo. MAVROS doesn't seem to be connecting to PX4. Therefore, no messages.

dagar, 2018-06-14:
No longer getting the early PX4 exit after reverting https://github.com/PX4/Firmware/commit/4eda0ed782faecdf600312cea7484e271d8008ea

https://github.com/PX4/Firmware/commits/pr-sitl_exit

lamping7, 2018-06-14:
I don't think this has to do with the shell. I think the problem is Jenkins workspace related. Our test is terminating PX4.

``` xml
<testsuite errors="0" failures="1" name="unittest.suite.TestSuite" tests="1" time="60.105">
<testcase classname="__main__.MavrosMissionTest" name="test_mission" time="60.1049">
<failure type="AssertionError">
failed to hear from all subscribed simulation topics | topic ready flags: {'mission_wp': False, 'ext_state': False, 'global_pos': False, 'state': False, 'alt': False, 'local_pos': False, 'home_pos': False} | timeout(seconds): 60 File "/usr/lib/python2.7/unittest/case.py", line 329, in run testMethod() File "/tmp/jenkins/workspace/Firmware_PR-9673-P47FTOH7TDQAC4XTOP6F6YLYP6VKKSMUMRJJ3IEVNXEO5335QLXQ/px4-posix_sitl_default-c1ec569/px4/integrationtests/python_src/px4_it/mavros/mission_test.py", line 247, in test_mission self.wait_for_topics(60) File "/tmp/jenkins/workspace/Firmware_PR-9673-P47FTOH7TDQAC4XTOP6F6YLYP6VKKSMUMRJJ3IEVNXEO5335QLXQ/px4-posix_sitl_default-c1ec569/px4/integrationtests/python_src/px4_it/mavros/mavros_test_common.py", line 248, in wait_for_topics format(self.sub_topics_ready, timeout))) File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue raise self.failureException(msg)
</failure>
```

The same issue (it seems like to me) is causing other failures too.
[image]

bkueng, 2018-06-14:
> I don't think this has to do with the shell. I think the problem could be Jenkins workspace related. Our test is terminating PX4.

Good, I thought it was strange too that it could be the shell (and if it were, it should always fail).

dagar, 2018-06-14:
> I don't think this has to do with the shell. I think the problem could be Jenkins workspace related. Our test is terminating PX4.

I'm not following, there are multiple failures in CI at the moment. The output I posted above is almost certainly from the EOF shell change, but running non-interactively sounds like a good fix.

lamping7, 2018-06-14:
Hmmmm.... I'm beginning to think you maybe right and the test failure is a result of px4 shutting down... 

For the second failure type:
It is possible for a workspace to have multiple packages. They get extracted. Wildcard completion everywhere makes things non-deterministic. The branch I'm working on makes all this happen inside the docker container -- which gets tossed in the trash after each stage/test -- rather than the workspace, which is shared, so we can't have this problem. It also saves disk space on the agents.

LorenzMeier, 2018-06-14:
I think that’s important. I really don’t like that things are staying after the test - that’s just always going to be an issue.

dagar, 2018-06-15:
Fixed.

closed 2018-06-15 by dagar