PX4/PX4-Autopilot #9691: Land (Navigator) for multi-copter

dakejahl, 2018-06-15:
_The vehicle will land at the location at which the mode was engaged. The vehicle descends at the rate specified in MPC_LAND_SPEED until it hits the ground._

A quad will attempt to land at the position it was at when this command was issued. If the quad is moving when this command is issued, it will overshoot this position (magnitude of overshoot depending on the inertia) and have to fly back towards it (all the while descending). This will often result in the drone touching down with considerable X,Y velocity, often resulting in a flip.

Heres a log: https://review.px4.io/plot_app?log=f323ad57-98ac-4372-808e-32a5f862ff8a

I think this can be improved fairly easily with some logic in the on_active() function in _/navigator/land.cpp_. Either an arrest of the horizontal velocity before descent, or simply requiring first an X/Y position change followed by a descent (similar to RTL). I am thinking about this in the context of a quad copter, so I am not sure of the implications for a fixed wing aircraft.

Does this sound like a reasonable thing to do? Are there any reasons this is a bad idea? I'd really appreciate some feedback.

Thanks,
Jake 

[image]
[image]

dagar, 2018-06-15:
This sounds like a great thing to do.

How about expanding Navigator LAND to have 2 simple states.
  1. brake (bypass for FW)
  2. land

The missing piece is then how navigator asks for the full stop.

@Stifael @MaEtUgR what do you think?

Stifael, 2018-06-16:
It is not just for land required, but also for other navigator modes. For instance, currently the vehicle overshoots when being first in fast forward flight with transition to rtl: the rtl climb lat/lon is the location at the time rtl was initiated.
In the past I pushed a PR that added a break item (https://github.com/PX4/Firmware/pull/8057/files). It basically asks consists of a waypoint that asks for 0 velocity.

BazookaJoe1900, 2018-06-16:
@Stifael, It might also related to https://github.com/PX4/Firmware/issues/7567 (Inconsistent RTL state) where in RTL there is big overshoot in the position, right?

dagar, 2018-06-16:
@Stifael what do you think about a new position setpoint type for BRAKE or ALL STOP?

Stifael, 2018-06-18:
@BazookaJoe1900, #7567 should not be related. #7567 is a bug, this issue here is just a missing implementation where the actual physics of the vehicle are neglected (it is assumed that the vehicle has infinite acceleration capability).

@dagar new type is one solution. However, not sure if it agrees with the Navigator triplet API, since all it is needed is just a type.

dakejahl, 2018-06-26:
Please see #9772

closed 2018-08-17 by dakejahl