PX4/PX4-Autopilot #9696: Poor land detection

LorenzMeier, 2018-06-16:
@MaEtUgR @bkueng Could you have a look? This is blocking v1.8

A new flight report just got uploaded:
https://review.px4.io/plot_app?log=e65ab2c6-18e9-408f-b442-4f19a41acf83

Description: beta18_mc02_aer(01)
Feedback: auto take-off; major loss of altitude between waypoints; poor land detection
Rating: Unsatisfactory
Wind Speed: Calm
Uploader: santiago@3dr.com

Vehicle type: Quadrotor
Airframe: Intel Aero Ready to Fly Drone, Quadrotor x
Hardware: AEROFC_V1
Vehicle UUID: 004500403335511839343335
Software git hash: aa625d9af880e3be7bd2ddd7ed49d10c74c6ada4

bkueng, 2018-06-18:
@santiago3dr Can you describe what you mean with poor land detection?

MaEtUgR, 2018-06-18:
The first suspect for any problems with land detection is in my experience the **hover thrust** configuration.
In the log `MPC_THR_HOVER, 0.5` the hover thrust is 50% while the vehicle uses **~38%** to hover in the equilibrium. Not sure from the problem description but it might be that **setting the parameter `MPC_THR_HOVER` to 0.38** already solves the problem or at least helps to improve it.

santiago3dr, 2018-06-18:
@bkueng 
The issue I encounter is that when the copter touches the ground, sometimes the copter jumps back up about 1 meter and lands again or the other thing that happens is the copter upon touching the ground, lifts two legs and sort of jumps a few times before disarming.

I was not able to reproduce this in today's flights but as soon as I see it again, I'll send you a video clip of said behavior.

santiago3dr, 2018-06-19:
I was able to reproduce the issue with the new stable release 1.8.0

great land detection:
https://review.px4.io/plot_app?log=9e3e76c7-7ac5-4e17-a624-069706e0d654

poor land detection:
https://logs.px4.io/plot_app?log=406113d2-d981-4794-958f-5fb6b295a2ec

video:
https://drive.google.com/open?id=1ad_aCtLa5KU97Y7FDC6LLEK0hgDC8m2v

PR #11172 (2019-01-26): Slow landing improvement suggestion
**Context**
Search for "landing" after this comment: https://github.com/PX4/Firmware/pull/9731#issuecomment-403600121
hopefully fixes #9696
fixes #10732
hopefully fixes #10001

**Test data / coverage**
SITL still works. Release testing for the upcoming H520 update shows a massive speedup in land detection.

**Describe problem solved by the proposed pull request**
When introducing the flight tasks architecture almost the whole legacy position controller was replaced a lot of things were modularized and simplified from a developers perspective. One of the thin [... cut]
closed 2019-01-26 by MaEtUgR